someweisguy/esp_dmx #1: Unable to build

AndrewBHarmon, 2021-07-06:
I cloned this into the components folder of my project and attempted to build. I am getting an error 
`` ../components/esp_dmx/include/dmx_caps.h:7:10: fatal error: soc/uart_caps.h: No such file or directory
``

I am unable to find a file ``soc/uart_caps.h`` either on my local install of ESP-IDF or on Espressifs Github. Does this library require a specific version of ESP-IDF? Am I missing a setup step? Do I need to create this file?

For context:
I am attempting to get DMX up and running on an existing project with other functionalities. I was able to build and flash before attempting to add this library. I am using ESP-IDF version ``v4.4-dev-1404-gc13afea63`` on Linux.

someweisguy, 2021-07-06:
Thanks for the feedback.

This is occurring because a dummy mistake that I made when writing the library. The `soc/uart_caps.h` file that is referenced is part of the [Hardware Abstraction Layer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/hardware-abstraction.html). The documentation warns not to use the files in the HAL because they are subject to change at any time, but I ignored it and included those files anyway. 

I'm currently working on an update that removes those dependencies. This is currently priority number one for me. In the meantime, the latest IDF build that I've tested with this library is v4.2.1. If you switch to that branch of the IDF, this error will go away. I have some free time in about two hours and I'll give v4.3 a quick test and let you know.

someweisguy, 2021-07-06:
It looks like this error also exists on ESP-IDF v4.3 which is disappointing. 

Let me know if you have any other thoughts. In the meantime, I'll get to work on fixing those dependencies. Thanks so much for bringing this to my attention! :)

AndrewBHarmon, 2021-07-07:
Awesome! Thanks for your help with this!

someweisguy, 2021-07-11:
It appears we can successfully receive DMX again. I've created a branch called `arduino_rx` because, incidentally, my primary objective with the update is to make this library Arduino compatible. It just so happens that the process of porting to Arduino fixes the linking errors!

Transmitting DMX works, but there appears to be an issue where the initial frame that is transmitted is garbage data. I will be working on this on the `dev` branch. If you only need to receive DMX on ESP-IDF v4.3 or greater, feel free to try the `arduino_rx` branch.

I will update you when I make that garbage data bug go away! I am pretty sure I know what is going on, so hopefully that will be quick - maybe today.

someweisguy, 2021-07-11:
Ah, yep. It appears I fixed the garbage data issue as well. Give the `dev` branch a try.

someweisguy, 2021-07-17:
Changes have been merged to the release branch. Thanks!

closed 2021-07-17 by someweisguy