someweisguy/esp_dmx #2: How to use with Platofrmio / Arduino Frameform

ellogwen, 2021-08-31:
Hi, I am trying for hours to get this to work with Platform.IO and the Arduino Framework.
Do you know how I could mix this esp-idf component withing my existing Arduino project?

ellogwen, 2021-09-02:
Hi, thanks for your answer. No need to rush or stress yourself :) I am currently using a modified LXESP32DMX to work with uart1, but this library needs to patch a core file for receiving to work, which is nothing I can do easily, so I am using it only for transmitting. For now. I will switch definetly in the near future.

I would offer you my help, but I have no clue about esp-idf though.

someweisguy, 2021-09-10:
I've pushed some code to the `dev` branch that builds on Arduino! Woohoo! You can clone that branch and copy the folder into your `libraries` folder in your Arduino directory, and that should allow you to `#include "esp_dmx.h"`.

I am doubtful that the keyword highlighting will work, but I am making progress. I am not thrilled with the fixes I made in the last commit; I'm still trying to figure out how the ESP32 Arduino core works in the grand scheme of things so I am sure that there are better ways to do what I did. For now, this code will work on ESP32, but it can likely break on ESP32-S2 or ESP32-C3. If you use the latter, let me know and I can push a hotfix so that you will be able to play with this library.

No word on Platform.IO yet. That will probably be much, much later.

someweisguy, 2021-09-10:
Highlighting works now! Turns out I had to use a single tab instead of spaces in my `keywords.txt` file. Whoops!

someweisguy, 2021-09-11:
The latest commit did some reorganizing. In `/examples` I've renamed all the example files. The example files for ESP-IDF code are now prefixed with `idf_`. I'm not sure if I love that formatting so it may change in the future. 

I've also added some [Arduino DMX read example code](https://github.com/someweisguy/esp_dmx/tree/dev/examples/read)!

Finally, I found an interesting bug in the previous code. Whenever you called `dmx_driver_delete()` the ESP32 would attempt to free the heap-allocated double-buffer. It would then detect a corrupted heap and would crash. I couldn't find any other solutions online, but what I found is that when the size of the DMX double-buffer is divisible by 4, the problem goes away. I suspect what is happening is that the Arduino implementation of ESP32's `malloc()` function only allows you to allocate 32-bit aligned memory as if calling `heap_caps_malloc(size, MALLOC_CAP_32BIT)` in ESP-IDF. Long story short, only allocate memory by factors of 4 on ESP32 Arduino!

someweisguy, 2021-09-11:
Another update: I've finished adding example code for Arduino! I'm going to style those examples so that they better fit the Arduino example sketch style guide, but they work now as is.

I found that Arduino-ESP32 v1.0.6 and earlier don't support the newest ESP32 chipsets. Arduino-ESP32 v2.0.0 was JUST released less than two weeks ago, and it does support the S2 and the C3. For the sake of inclusion, I'm going to support Arduino-ESP32 v2.0.0 and up only. If you want to use Arduino v1.0.6, I've created a branch called `arduino-v1.0.6` but I'm not planning on updating it in the future.

someweisguy, 2021-09-12:
I've officially ported this library to Arduino. It's available in the Arduino IDE library manager and the example code is working. So I am going to close this issue. 

I'm still looking into Platform.IO. It looks simple enough, but that will be another release! Let me know if you have any other questions. :)

closed 2021-09-12 by someweisguy