someweisguy/esp_dmx #3: VSCode and Platformio

berlinrob, 2021-09-23:
Great work Mitch, thank you.
This is not an issue with the current code and IDE, but related to another IDE. I tried to run this project from VSCode with Platformio and I get some errors for missing datatypes (like uart_sclk_t). In your instructions you are clear about using the latest framework. Because I am using the arduino framework I cannot use the latest espidf framework which is certainly causing the missing datatype errors. Have you used VSCode with Platformio successfully with your project?

someweisguy, 2021-09-25:
Thanks for the question and the kind words!

I took a brief look at this and I think I see what is happening. I _believe_ that PlatformIO doesn't use Arduino-ESP32 v2.0.0 yet because it is so new. When you use Arduino-ESP32 v1.0.6, you get these errors, such as the `uart_sclk_t` not defined. These symbols are defined in v2.0.0 so the error goes away. 

There is a [branch for Arduino-ESP32 v1.0.6](https://github.com/someweisguy/esp_dmx/tree/arduino-v1.0.6) that solves these problems, but I haven't set that branch up for use with PlatformIO which is why I believe you are getting the error. If you can, you can try cloning that branch into your project. I believe the error will go away. 

Sorry, this was a silly error on my part. It should be an easy fix, but the in-laws are in town this weekend, so I am not sure if I will have time to work on it for a few days.

berlinrob, 2021-09-25:
I just did a quick test with the dmx_read.ino from the branch you mentioned. It successfully compiled and ran using PlatformIO. 

In my test project I modified the code in the dmx_read.ino and included the Adafruit_NeoPixel.h. Basically the mod is just changing a neopixel RGB colors based on the dmx values received from three sliders on a dmx controller board. Thank you for that tip!!!

Honor the in-laws and perhaps we talk again soon.

someweisguy, 2021-10-02:
Thanks for you patience on this. Weekend with the in-laws was great, then this past week I opened a theater show, so it has been busy. I'm glad you got it working! That is great to hear and definitely a load off my mind. 

To expand a bit on the errors you received, the ESP32-Arduino API didn't expose certain symbols that allow for you to target the latest ESP32 chipsets. I believe one of these symbols used is `SOC_UART_NUM`. It describes the number of UARTs the chipset has, so the library uses that symbol to throw errors if, say, the user tries to initialize UART 2 (the third UART) on the ESP32-S2. This is won't throw an error on ESP32 but because the S2 only has 2 UARTs it needs to throw an error on the S2. There are more useful symbols that aren't defined (like the `uart_sclk_t` which is used to define which hardware clock to use with your UART), but `SOC_UART_NUM` is the main one. Since ESP32-Arduino v2.0.0 was just recently released, PlatformIO still uses v1.0.6 which doesn't expose those useful chip-targeting symbols. 

All this to say, this should be a simple fix. I basically have to conditionally define certain symbols depending on which version of Arduino we are using.

someweisguy, 2021-10-02:
Actually, I think I may be onto an even simpler fix. You can define which version of Arduino-ESP32 you would like to use in your `platformio.ini` file. I've added the following lines to it: 

```
platform_packages=
   toolchain-xtensa32 @ ~2.80400.0
   framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.0
```

When you add this (specifically the last line) PIO will install the entire Arduino-ESP32 source from its repository, so PIO will tell you it's working on tasks for a while. Took me about 10 minutes on my ~100Mbps/down connection. With these changes, it builds now, but it is not linking properly. I'll do some more investigation.

someweisguy, 2021-10-02:
I've found the solution based on [this issue](https://github.com/espressif/arduino-esp32/issues/5436#issuecomment-885469600)! I've updated my `platformio.ini` file to use an updated platform. Instead of `platform = espressif32`, I use `platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream`. The first part of that platform is the link to the PIO ESP32 repository, and the `#feature/arduino-upstream` tells PIO to use the `arduino-upstream` branch. 

All together, this is what my `platformio.ini` file looks like;

```
[env:featheresp32]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = featheresp32
framework = arduino
lib_deps = someweisguy/esp_dmx@^1.1.1
```

Like my earlier comment, it takes about 10 minutes of PIO running tasks to install the new platform. When it finish, the library builds and links. I still need to test the example code, but once I do and everything works, I'll update the readme for instructions on how to solve this and then close this issue. Let me know if you have any questions in the meantime!