someweisguy/esp_dmx #5: Serial Monitor output is only the character "null"

quirsh, 2021-10-04:
Hi.

First of all many thanks for this great library you provided. It looks really awesome!

I tried to run your "DMXWrite" example on my machine but unfortuantely the serial monitor output is just garbage. I use VSCodium and PlatformIO and your instructions in the README were very helpful. 
I checked that I use the right monitor speed (115200) but still I only get the "null" character (U+2400) back.
I traced it back to line 43 "dmx_param_config(dmxPort, &dmxConfig);" as if I remove it (and all of the depending code) the problem does not appear anymore. Not sure if this is helpful as I'm also overwhelmed with the internal clockworks of your library.

Any help would be very appreciated.

Thanks, quirsh

someweisguy, 2021-10-04:
Thanks for the question and the kind words!

On line 43, when you call `dmx_param_config()`, what is the value of `dmxPort`? Is it `0`, `1`, or `2`? `dmx_param_config()` configures the UART hardware for use with DMX. The serial monitor uses UART 0, so if the value of `dmxPort` is 0, that could cause the issue that you are experiencing.

quirsh, 2021-10-05:
I used the default which is 2. I also tried with the two other ports but nothing changed.
I was physically able to verify that the DMX output works and the serial port pin is not influenced by it (continous data stream on DMX pin, interrupted data stream on serial port pin, fits with the incoming charcters).

(PS: Just realized that I gave the wrong line number, it's 54 in the example file, my bad)

someweisguy, 2021-10-05:
Thanks for the info. I'm glad to hear that the DMX output is working. I just ran the `DMXWrite.ino` example on my ESP32 Feather last night and I didn't have any issues, so I'm wondering if you are using a different ESP32 board? Which one are you using?

Can you also share what your `platformio.ini` file looks like?

someweisguy, 2021-10-06:
I want to say that this has something to do with how your UART is being configured in PlatformIO, but that doesn't explain why the problem goes away when you get rid of `dmx_param_config()`.

Take a look at [this line](https://github.com/someweisguy/esp_dmx/blob/e547de8f245ebcbc368212d312be9f9ad7007247/src/esp_dmx.c#L16) in `esp_dmx.c`. It might be worth deleting lines 16 through 26 inclusive and replacing with `#define CONSOLE_UART_NUM 0`. I think that for some reason, `dmx_param_config()` is [disabling the UART peripheral at this line](https://github.com/someweisguy/esp_dmx/blob/e547de8f245ebcbc368212d312be9f9ad7007247/src/esp_dmx.c#L201), causing the null characters.

quirsh, 2021-10-06:
The suggested change made the output a bit more diverse but I can't really say if it goes in the right direction. Today I also tried it with an other physical board (same model) to be sure it's not a faulty one.

I might be going in the wrong direction, but is there a way to figure out which datastream is running on which UART port? They aren't hardwired to the physical pins, are they?

Serial output, same options, still the Write example: 
```
e�sZ␘�R␀␀␅␀␀␀␀E␀"␐␀␀␌(␀␂␐&␇j����N�E�␀␆H␀␒␀�␀b␁␀ �Z␂␄␀�␀0␀␀␅␀0␄�␄␀K␀ �!� ␀␀�`��C␞␂␀�␀␁V␀ �␆��␀E␀␀Ԅ␕␂␁␀␀␁␁␀␀ ␌␀␀␀�␀␀Z%�
,␒0␂Ȕ� �v␄␀␀␀␀␀
X1C�2␁�XE0�Qd0␐␀ȓ\␀
?␄␂�␐�␔H␌�␂␀␀␀�␀�␀P�␂   �␁T ␅␀P␀�
␀␃�␎K␑�␀�␀␀@�␁"a�␀�␀�␀<␏␀eN␃␀�␀�Đ2ɤ␕␁�␀]␖Q␒D��␀␑␀␀�␆A    �␁r@␀�
␀␄␀␀␀M␃␀␄␀␂�␎␀@␀␘�␀␅���␆␀d␀␀␂␀␂␁␄t␀␄␀ ␀@@␁␅��ȩ ␀�␀X␀jA␀� ␀ �␀␕␀␁␀␀␀␋␀␀a0z␁6�P:␁��␎␀␀␇␀␀␀␀␆ɖa�T�␄␓�$␀0�␀␀␁�␁␄�␀�␏��␀�␀@␀d␐@␐␁ �␋␀␀␀��␀␀␃op␀�␀␀␆␂␀*�H␀␌␂(␀ 
```

someweisguy, 2021-10-07:
I've tested out a different board. I used an ESP32-DevKitC, which is an ES32-WROVER-E, but that shouldn't make a difference. Same chip with different amount of RAM. I built and uploaded the `DMXWrite.ino` example file as-is, and I copy/pasted your `platformio.ini` to mine. I wasn't able to get it to build due to some dependency issues with ottowinter's libraries, but I removed those and it uploaded fine. So I am more or less using the same system you are using, but I am not able to reproduce your error. Grrr....

I am not sure there is great way to see which UARTs are being used for what. UART 0 is used for the serial monitor in all boards that I am aware of. UART 0 is transmitted on GPIO 1 and 3. UART 1 and 2 are free for use for whatever is wanted, and you can assign them to almost any other GPIO on the board. When I had you replace those lines with `#define CONSOLE_UART_NUM 0`, my intention was to test to see whether or not we were clobbering your serial monitor on UART 0 by accidentally setting it to DMX speeds. I feel pretty confident that isn't what is happening here, but when you think about what `dmx_param_config()` does, it is hard to imagine what else is going on. 

Here's what `dmx_param_config()` in `esp_dmx.c` does:
1. [First, we do some error checking to make sure that the user didn't pass arguments that don't make sense.](https://github.com/someweisguy/esp_dmx/blob/e547de8f245ebcbc368212d312be9f9ad7007247/src/esp_dmx.c#L362-L383)
2. [Next, we enable the UART peripheral. If the UART selected is _not_ the serial monitor, we reset the peripheral first.](https://github.com/someweisguy/esp_dmx/blob/e547de8f245ebcbc368212d312be9f9ad7007247/src/esp_dmx.c#L385-L394) I said that it disables the UART in an earlier comment, but I linked you the wrong line. Sorry! It actually _enables_ the UART.
3. [Then we configure the UART hardware to use the correct data bits, parity, stop bits, etc. We also set break and MAB times.](https://github.com/someweisguy/esp_dmx/blob/e547de8f245ebcbc368212d312be9f9ad7007247/src/esp_dmx.c#L396-L403)
4. [Finally, we flush the UART hardware FIFOs so that we don't process any garbage data that may be in them.](https://github.com/someweisguy/esp_dmx/blob/e547de8f245ebcbc368212d312be9f9ad7007247/src/esp_dmx.c#L405-L407)

I'm going to keep chugging along on trying to recreate this problem, but I am stumped. You might try commenting out each of those sections individually and find out which fixes your serial problems. This would break the DMX, but it would at least give us a broad idea of what the issue is here. 

Definitely let me know if you have issues on Arduino IDE! Good thinking!

someweisguy, 2021-10-09:
> After some hassle getting ArduinoIDE to compile and upload sketches to ESP32 I was able to push the Write Example to my board and the serial monitor worked flawlessly. Not sure if this it good or bad :D

This leads me to believe there is something fishy going on with VSCodium and PlatformIO. I am running VSCode (not Codium) on PlatformIO. I tried to get VSCodium running, but I was not able to get the PIO extension installed on it. On VSCode, I cannot reproduce this error.

Going back to your earlier comment on seeing how each UART is configured, I wrote a little snippet that may help, which you can copy/paste to line 77 in the `DMXWrite.ino` example. It looks at the registers for the UART port that you specify and returns how each are configured. In this case, it should print `115201, 0, 1, 3` for UART 0 on the serial monitor. Since your serial monitor doesn't work, maybe try lighting up and LED or `digitalWrite()` a pin on the comment that suggests doing so?

```
// Change this to select different UART ports
int uart_port = 0;

uint32_t baud;
uart_parity_t parity;
uart_stop_bits_t stop_bits;
uart_word_length_t data_bits;

esp_err_t err = uart_get_baudrate(uart_port, &baud);
err |= uart_get_parity(uart_port, &parity);
err |= uart_get_stop_bits(uart_port, &stop_bits);
err |= uart_get_word_length(uart_port, &data_bits);

if (!err) {
  Serial.printf("%i, %i, %i, %i \n", baud, parity, stop_bits, data_bits);
  if ((baud >= 109440 && baud <= 120960) && parity == 0 && stop_bits == 1 && data_bits == 3) {
    // do something here? maybe light up an LED?
  }
}
```

If you can get this code to light up an LED for UART 0, this would confirm that the issue is with VSCodium/PlatformIO and not the code. An ESP-PROG would be a better idea for troubleshooting this, but that is a whole other can of worms.

someweisguy, 2021-10-12:
Thanks for the information about VSCodium and PIO - I will take a look!

Back to the problem at hand though: `data_bits == 3` is correct. On the ESP32, the number of data bits is is equal to `data_bits + 5`. So that would mean that you are transmitting 8-bit words, which is what we want. The data-bit/word-length constants for ESP32 can be found [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#_CPPv416UART_DATA_8_BITS) if you are interested.

Seeing `115200,8,N,1` is also what we want to see. It indicates that your PIO serial monitor is `115200` baud with `8` data bits, `N` for no parity, and `1` stop bit. `/dev/ttyUSB0` is the device that PIO is monitoring, which may not necessarily be accurate. Could you check to see what port you upload code to? An easy way to do this is to open the Arduino IDE and go to Tools->Port and see which option is selected. It may be something like `/dev/cu.SLAB_USBtoUART`.

martin3000, 2021-11-18:
Same problem here.
This is my code:

```
#include <esp_dmx.h>

void setup() {
  Serial.begin(115200);
  while (true) {
    delay(500);
    Serial.print("ok ");
  }
}

void loop() {
}

```
As you see, the problem arises by including esp_dmx.h
Including dmx_caps.h or dmx_types.h has the same effect.

someweisguy, 2021-11-19:
The issue was in `dmx/driver.h`. There was a struct named `uart_periph_signal`, whose name I copied verbatim from ESP-IDF in order to make the library compatible with Arduino. The Arduino framework hid a lot of useful ESP32 UART constants found in the IDF so I essentially copied them across in order to rewrite as little code as possible. To fix the bug, I renamed `uart_periph_signal` to `dmx_periph_signal`. I am guessing there was some sort of name conflict? Really strange.

In any case, I've pushed those changes to the `dev` branch. If you need to use Arduino-ESP32 v2.0.1, please clone the dev branch to your libraries folder. @quirsh I'm very interested to see if this fixes your problem. Please let me know if you are able to test out the changes!

martin3000, 2021-11-19:
Wow, great that you found it!!!

someweisguy, 2021-12-10:
Changes have been merged. Thank you!

closed 2021-12-10 by someweisguy

someweisguy, 2022-01-04:
Glad to hear it! Thanks for following up!