someweisguy/esp_dmx #7: arduino compiler error - type uart_sclk_t not found

theeprawn, 2021-10-30:
Using latest ardunio 1.8.15 and esp1.0.6 code
compiler error, type uart_sclk_t not found

readded your prevoius verson code to dmx_types.h file.

Now compiles, hopefully will be testing DMX output to uart2 soon.

---------------------------------------------------

// TODO: is this code defined somewhere in Arduino?
#ifdef ARDUINO
typedef enum {
    UART_SCLK_APB = 0x0,            // UART source clock from APB
    UART_SCLK_RTC = 0x1,            // UART source clock from RTC
    UART_SCLK_XTAL = 0x2,           // UART source clock from XTAL
    UART_SCLK_REF_TICK = 0x3,       // UART source clock from REF_TICK
} uart_sclk_t;
#endif

someweisguy, 2021-10-30:
Thanks for the feedback!

This library requires the latest Arduino-ESP32 framework which is v2.0.0, not v1.0.6. The reason for this is because v2.0.0 supports the ESP32, the S2, and the C3 whereas v1.0.6 does not.

For instructions on how to get the latest version of the Arduino-ESP32, check out the [readme](https://github.com/someweisguy/esp_dmx#arduino). Tutorials like [this one](https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/) are no good because they don't allow you to install v2.0.0.

Alternatively, if you really need to stay on Arduino-ESP32 v1.0.6, you can try cloning the [Arduino-v1.0.6 branch](https://github.com/someweisguy/esp_dmx/tree/arduino-v1.0.6) and using that instead.

closed 2021-10-30 by someweisguy

chaosloth, 2022-02-22:
Had same issue as OP -- Updated to 2.0.0 as per instructions and I am now able to compile and execute.  Thanks