someweisguy/esp_dmx #9: ESP32 core 2.0.1-RC1, UART0 does not work

pwolfman, 2021-12-10:
I have tested your DMX library on a ESP32 with core version 2.0.1-RC1 and it seems to work sending DMX packets on UART2 using your example code, expect UART0 stops working.  Below is the simplest code that fails.  If you comment out the #include ,esp_dmx.h> the below example will send "hello", if the library esp_dmx.h is included the serial port stops working.  I do get the initial ESP32 core bootup info and then nothing after that.   

#include <esp_dmx.h>
void setup() {
    Serial.begin(115200);
    Serial.println("setup serial");
}

void loop() {
    Serial.println("hello");
    delay(1000);
}

someweisguy, 2021-12-10:
Thanks for the feedback! Which version of esp_dmx are you using? This issue was reported [here](https://github.com/someweisguy/esp_dmx/issues/5#issuecomment-973236325) and should be fixed now in v1.1.3.

pwolfman, 2021-12-10:
I was using Ver 1.1.2 that I loaded 2 days ago:)  I just loaded 1.1.3 and a quick test shows it's working! I will do some more testing in the next few days.  Thanks for the support!

someweisguy, 2021-12-11:
Wahoo! Glad I could help! Please do open another issue if you find anything! :)

closed 2021-12-11 by someweisguy