someweisguy/esp_dmx #10: Random read input

PhilippDarkow1, 2021-12-13:
Hello,

first of all thank you for that nice library :) and i hope you can maybe help me with my problem. 

I am trying to read the dmx packets from a dmxcontroller (eurolight LED Color Chief) 

My current setup is shown below

[image]

I am using the DMXRead example and modified it a little bit. The good thing is that there is some input coming to the esp32 :) the bad thing is that it is random.

My setup function looks like this

```
void setup()
{
  /* Start the serial connection back to the computer so that we can log
    messages to the Serial Monitor. Lets set the baud rate to 115200. */
  Serial.begin(115200);
  Serial.println("Start Setup");

  readDMXTimer.setInterval(1000); // every second
  readDMXTimer.setCallback(checkDMXInput);

  /* Configure the DMX hardware to the default DMX settings and tell the DMX
    driver which hardware pins we are using. */
  dmx_config_t dmxConfig = DMX_DEFAULT_CONFIG;
  // const dmx_config_t dmxConfig = {
  //     .baud_rate = 250000, // 250000 typical baud rate
  //     .break_num = 45,     // 90, //45,     // 180us
  //     .idle_num = 5        // 5 //10         // 20us
  // };

  dmx_param_config(dmxPort, &dmxConfig);
  dmx_set_pin(dmxPort, transmitPin, receivePin, enablePin);

  /* Now we can install the DMX driver! We'll tell it which DMX port to use and
    how big our DMX packet is expected to be. We'll also pass to it our queue
    handle, some information about how many events we should store in our queue,
    and some interrupt priority information. Both the queue size and interrupt
    priority can be set to 1. */
  int queueSize = 10;        // 1;
  int interruptPriority = 1;
  dmx_driver_install(dmxPort, DMX_MAX_PACKET_SIZE, queueSize, &queue,
                     interruptPriority);

  Serial.println("End Setup");
}
```

I changed the queue size to 10 and i tried to use a custom dmx_config but switched it back to the default one. Also i use a timer to call the read function every second.

The implementation of the read function is 

```
void checkDMXInput()
{
  /* We need a place to store information about the DMX packet we receive. We
    will use a dmx_event_t to store that packet information.  */
  dmx_event_t event;

  /* And now we wait! The DMX standard defines the amount of time until DMX
    officially times out. That amount of time is converted into ESP32 clock ticks
    using the constant `DMX_RX_PACKET_TOUT_TICK`. If it takes longer than that
    amount of time to receive data, this if statement will evaluate to false. */
  if (xQueueReceive(queue, &packet, DMX_RX_PACKET_TOUT_TICK))
  {

    /* If this code gets called, it means we've recieved DMX data! */
    /* We should check to make sure that there weren't any DMX errors. */
    switch (event.status)
    {
    case DMX_OK:
      if (packet.size >= slot_idx)
      {
        dmx_read_packet(DMX_NUM_2, data, event.size);

        printf("Packet took %i microseconds!\n", event.duration);
         printf("Received packet with start code: %02X and size: %i\n",
                event.start_code, event.size);
      break;
    case DMX_ERR_IMPROPER_SLOT:
      printf("Received malformed byte at slot %i\n", event.size);
      // a slot in the packet is malformed - possibly a glitch due to the
      //  XLR connector? will need some more investigation
      // data can be recovered up until event.size
      break;

    case DMX_ERR_PACKET_SIZE:
      printf("Packet size %i is invalid\n", event.size);
      // the host DMX device is sending a bigger packet than it should
      // data may be recoverable but something went very wrong to get here
      break;

    case DMX_ERR_BUFFER_SIZE:
      printf("User DMX buffer is too small - received %i slots\n",
             event.size);
      // whoops - our buffer isn't big enough
      // this code will not run if buffer size is set to DMX_MAX_PACKET_SIZE
      break;
    case DMX_ERR_DATA_OVERFLOW:
      printf("Data could not be processed in time\n");
      // the UART FIFO overflowed
      // this could occur if the interrupt mask is misconfigured or if the
      //  DMX ISR is constantly preempted
      break;
    }
  }
  else if (dmxIsConnected)
  {
    Serial.println("DMX timed out !!!");
    Serial.println(event.status);
  }
  
  yield();
}
```

My output is 
```
Start Setup
End Setup
Received packet with start code: FC and size: 513
Received packet with start code: C0 and size: 2
Received packet with start code: FC and size: 513
Received packet with start code: 00 and size: 2
Received packet with start code: FC and size: 513
Received packet with start code: 00 and size: 2
Received packet with start code: FC and size: 513
Received packet with start code: 00 and size: 1
Packet size 0 is invalid
Received packet with start code: FC and size: 513
Received packet with start code: 00 and size: 1
Received packet with start code: FC and size: 513
Received packet with start code: FE and size: 513
Packet size 4658 is invalid
Packet size 4652 is invalid
Packet size 1033 is invalid
Packet size 1033 is invalid
Packet size 1033 is invalid
Packet size 1033 is invalid
Received packet with start code: FC and size: 513
Received packet with start code: FE and size: 513
Received packet with start code: 00 and size: 513
Received packet with start code: 00 and size: 8
Packet size 1032 is invalid
Received packet with start code: FC and size: 513
Received packet with start code: FC and size: 513
Received packet with start code: FC and size: 513
Received packet with start code: FC and size: 513
Received packet with start code: FC and size: 513
```

Do you have maybe an idea what I am doing wrong?

Thank you in advance

someweisguy, 2021-12-14:
Thanks for the feedback! Those RS485 breakout board are tricky. I remember trying to get them working when I was first developing this library, and I couldn't for the life of me figure out what I was doing wrong. My solution was to make my own PCB from scratch.

That said, it looks like you are supplying your breakout board with 3.3v. The [MAX485](https://datasheets.maximintegrated.com/en/ds/MAX1487-MAX491.pdf) takes 5v Vcc. I would try switching over to the 5v rail of your ESP32 and see if that fixes the issue.

PhilippDarkow1, 2021-12-14:
Thank you that was the issue with the 5v volt. I can read the data now :)

closed 2021-12-14 by PhilippDarkow1

meczero, 2025-10-31:
@someweisguy almost 4 years and I need to say thank you as I have the same issue but with a total different project and it was because vcc on that dam max485 😄