someweisguy/esp_dmx #11: Can this library be used to connect an ESP-32 to a DMX wireless circuit board?

BrainFreezeSoftware, 2021-12-31:
I'd like to connect an ESP-32 development board to a DMX wireless circuit board like this:

https://www.amazon.com/Wireless-Transceiver-Controller-Transmitter-Receiver/dp/B07GZ28DFL/ref=sr_1_73?keywords=dmx+wireless+receiver+transmitter&qid=1640913708&sprefix=dmx+w%2Caps%2C65&sr=8-73

If it can, any idea of wiring would also be helpful.

Thanks,
Steve

someweisguy, 2021-12-31:
It looks like it is possible. To do so, you would want to attach a RS485 or DMX interface (like [this one](https://www.sparkfun.com/products/15110)) to your ESP32, and then plug XLR pin 1, 2, and 3 into G, D-, and D+ inputs of your product, respectively. This would be the surest way to get this product working with the ESP32. 

It looks like your product has a TX, RX, and Ground pin for a UART as well. You _may_ be able to plug your ESP32 UART into those pins to transmit DMX but it isn't very clear if that is possible. It may just be a programming port for the product. If you are able to transmit DMX via straight UART, you'd likely need a logic level converter to convert the 3.3v ESP32 UART into 5v.

Any sort of wireless communication from the ESP32 to your product is not possible with this library. 

Let me know if you have any other questions!

BrainFreezeSoftware, 2021-12-31:
@someweisguy Thanks for the response. I really appreciate it.

I looked at the Sparkfun board, but it's overkill for what I'm looking to build. Really trying to tap into wireless DMX. Sounds like the RS485 is the way to go. Something like this?

https://www.amazon.com/ALMOCN-Adapter-Module-Converter-Indicator/dp/B08XLT21S6/ref=sr_1_9?crid=1XIQEPYN3VBN1&keywords=RS485&qid=1640989381&sprefix=rs485%2Caps%2C99&sr=8-9

In your sample code you talk about RTS: IO21. What is it used for and does it need to be attached somewhere?

I think I understand what you're saying about how it would be connected. I'll probably get some parts and give it a try. See what happens.

someweisguy, 2022-01-01:
That will work!

The way that RS485 works (and DMX too, since DMX uses RS485) is that it sends data using the D+ and D- lines in a half duplex data bus. Half duplex means that it can only send data or receive data, but not both at the same time. Therefore, it is necessary to signal to your RS485 chip when you want to receive data and when you want to send data. That is what IO21 is used for. RTS stands for "Request To Send." Typically, you'll pull the RTS line low and that tells your RS485 chip that it should receive data, not send it. If RTS is high, it tells the RS485 chip that you are transmitting data. Often you'll see DE/RE instead of RTS - it stands for Driver Enable/Read Enable.

It looks like the product you linked doesn't have a RTS pin. It is possible that its RTS logic is automatic. There RS485 boards that will stay in "read" mode until it detects data on the TX line. When it does, it switches to "write" mode and transmits your data. I am guessing that is the case here.

Since you aren't using that pin, your [setting communication pins](https://github.com/someweisguy/esp_dmx#setting-communication-pins) step will look a bit different. You can call `DMX_PIN_NO_CHANGE` when you declare your RTS pin. Like this:

```
const int dmx_num = 2;
const int tx_io_num = 17, rx_io_num = 16;
dmx_set_pin(dmx_num, tx_io_num, rx_io_num, DMX_PIN_NO_CHANGE);
```

Let me know if you have any other questions! :)

closed 2022-01-03 by someweisguy