someweisguy/esp_dmx #15: pin assignment for ESP32 & 485

nada0302, 2022-04-03:
Thanks for nice library.
I tried using example as it is [DMXRead](https://github.com/someweisguy/esp_dmx/tree/release/v1.1/examples/DMXRead)

but cannot detect DMX signal and found cannot go through both following if statement ...

- if (xQueueReceive(queue, &packet, DMX_RX_PACKET_TOUT_TICK)) {
- } else if (dmxIsConnected) {

pin diagram is followed by here 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#circuit-a-collision-detection-circuit

but, is my wire connection wrong ?
[image]

someweisguy, 2022-04-05:
Thanks for the kind words!

These RS485 circuits can be tricky sometimes and you shouldn't need to make any big changes to the example code. Your wiring diagram looks correct to me, but it's been a while since I've had to think about how the wiring in those RS485 breakout boards are supposed to look.

Does your RS485 breakout board require 5v data?

nada0302, 2022-04-13:
Thanks for comment. I will check in the weekends and update.

nada0302, 2022-04-17:
I couldn't solve, but I will try later and may come back. Thks,

nada0302, 2022-05-08:
Finally, I found my mistake.
Wire layout OK. 
Just wrong to use ESP32-WROVER, which is not able to use pin 16&17.
VCC should be better for 3.3V as ESP32 GPIO is by 3.3V basis.

someweisguy, 2022-05-11:
Ah, that is a good find! I'm going to reopen this so that documentation can be updated, since all the examples use those pins. Hopefully we can prevent confusion in the future.

someweisguy, 2022-05-22:
I've updated the readme on the `dev` branch. Hopefully, it is helpful for building RS-485 circuits! I'll close this when I've published a new release.

MnemonicFish, 2022-05-26:
ESP32 won't work with that module, it uses 5v logic, a RS3485 runs 3.3v logic. The Youmile RS485 module operates with 3.3v and 5v, works with ESP32 by connecting its VCC to 3.3v not 5v.

https://www.amazon.co.uk/gp/product/B08BZ8274M/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

nada0302, 2022-05-30:
I confirmed run by the module, attached on the first photo. (now I use M5 stack core basic for check, which is ESP32-D0WDQ6-V3) , VCC by 3.3V

someweisguy, 2022-06-07:
v2.0 is released and addresses these issues. Thanks so much!

closed 2022-06-07 by someweisguy