someweisguy/esp_dmx #16: DMX RDM

midikidi, 2022-04-12:
Hi has anyone managed to implement RDM with this great library

someweisguy, 2022-04-13:
Thanks for the kind words!

To my knowledge, no, but it should be possible to do so. I have been wanting to give it a shot, but I don't have any RDM capable lights laying around. :/

I also think this library falls on a lower level of the OSI model, so if/when I ever get RDM working, I think it would be its own separate library that wraps this library.

midikidi, 2022-04-13:
I m currently porting a few other librarys but once finished I will give it a go. I will keep you informed if i have issues i will let you know looking at either dmxserial2 as a staring point or esp32dmx
Dave

ebeam-bob, 2023-02-16:
Would it be possible to re-open this, or should I make a new one? I've made good progress towards an RDM responder but have run into an issue.  TIA, Bob

someweisguy, 2023-02-17:
I've reopened this! I'd love to hear what you are working on. 

I've been taking a look at the ESP event loop library. It looks promising, but I think it'd need to be modified a bit in order to make RDM responders work. I haven't had much time to work on it because I've been fixing bugs in the meantime.

ebeam-bob, 2023-02-17:
Hi Mitch:

Thanks for reopening the issue.  Your esp_dmx library is in the best shape
of all the ones I've tried (3-4), and you left me enough breadcrumbs to get
the responder side of discovery working, and from there, once I had a
working version of rdm_send_response(), I was off to the races!

I'm working on a "real" version of the ESP32 & SparkFun DMX-to-LED shield
for outdoor/architectural LED lighting, and I have a "professional" Swisson
XMT-350 DMX/RDM tester that shows that the discovery and get/set code,
based on your library, work correctly.

The problem I have is that I can go into RDM mode once after my code starts
up, and it responds to the Swisson with all the ID strings I put in the
device info, etc.  But if I go out of RDM and "back" to DMX, the DMX code
still updates the LED strings properly, but I can't go back into RDM.  The
Driver Enable to the RS-485 transceiver is not going high like it
did before, even though the library is flagging the packets as RDM.

Any suggestions?
TIA,
Bob

someweisguy, 2023-02-17:
Thanks for the kind words!

It's funny - I've always wanted to purchase one of those Swisson DMX/RDM testers, but I didn't want to spend $500USD on one. So instead I spent two years developing this library so that I could DIY my own! I have a cheap logic analyzer and oscilloscope that I use for testing this library to make sure it works. I also have a few RDM compatible fixtures at work that I can occasionally test with. Still, I'm so thrilled to hear that the Swisson is able to understand the DMX from this library!

To paraphrase your issue, it sounds like your ESP32 will respond to the first RDM message it receives, but doesn't respond to messages afterwards. Off the top of my head, I am not sure what could be causing this. If you have example code that you could send me, that would be helpful.

One thing that I would check is that, as a part of the RDM Discovery process, RDM devices are muted. This meant that they will not respond to RDM Discovery messages until after they are unmuted. The RDM Discovery Mute is only applicable to RDM Discovery messages. If, for example, your RDM responder is muted and you send it a `IDENTIFY_DEVICE` message, your RDM responder must respond. If your RDM responder is muted and you send it a `DISC_UNIQUE_BRANCH` message, it will not respond.

ebeam-bob, 2023-02-17:
A Swisson is a great tool and if I could loan you mine, I would... Is it possible to contact you outside Github? I followed you on Twitter as a possibility.

You have the crux of the issue there.  I've had to modify esp_rdm.c and some of the private files so I should probably share those with you.  Can you suggest how I should do that?

I just checked my RDM_PID_DISC_MUTE code and it does call set_rdm_muted() as does RDM_PID_DISC_UN_MUTE (args of true and false, respectively).  I'll dig into this a little deeper and see if the initial setting of "muted" has anything to do with this.  I'll also take a look at the Destination UID qualification I use, since I think the Swisson should be doing a broadcast un-mute at the beginning of discovery and if I'm not letting that through the second time, that might be the problem.

someweisguy, 2023-02-19:
Absolutely. You are welcome to reach out to me on Twitter. I'm also on Discord as someweisguy#2556.

For sharing code, I'd say using git is your best bet. You are able to fork this repository to make your own changes. After forking a repository it is then possible to submit a pull request to merge your changes with the main codebase. If you're not familiar with git, it can be a massive pain to learn for the first time. But after you start to understand it it is incredibly helpful. 

If you are already familiar with git, then I apologize for explaining what you already know! :)

ebeam-bob, 2023-02-24:
I just put in a Pull Request after checking to see what changed on my side.  I also noticed that esp_dmx.c changed quite a bit from the version I'm using, so as soon as I can (we're in the middle of getting some beta sites set up so I need to support them) I'll see if that fixes anything on my responder.

bethlenke, 2023-03-12:
I also tried to implement RMD as well bit I stuck, I have a DMXCat tester hooked up, but I was unable to make my esp discoverable by it. 

I might be stupid for this one, but if any of you get some time on your hands can you help me to the point where you guys are? :)

also, I have a RDM compatible fixture, so if needed I can provide captured data!

thx, bethlenke

someweisguy, 2023-03-16:
Sorry for the late response on this! I've been busy lately, so I haven't had as much time to work on RDM as I hoped.

I did have some time to investigate how the RDM callbacks would fire. I first tried setting up a FreeRTOS task whose purpose is to respond to RDM packets. I found that this wouldn't work because this library uses a mutex in all of its functions. If the mutex was held by a task other than the RDM responder task, it would mean that there is a potential for up to 10ms delay before the RDM responder task would be able to respond. If I recall correctly, RDM responders must respond to packets within 2.8ms. So the RDM responder task solution wouldn't work.

I think the best way forward is to have the RDM response function fire in the `dmx_receive()` function. This would work very similarly to @arneboe's solution. The user would be able to register callbacks for RDM responses. The RDM response function would search a dispatch table for the appropriate callback (if it exists) and execute the callback. Data from the callback would then be passed back to the RDM response function and the ESP32 would send a response packet.

netmindz, 2023-03-23:
@arneboe - you might be interested to know that I'm in the process of using this library to add "classic" DMX support to WLED to behave just as the DMX over IP options do https://github.com/netmindz/WLED/tree/DMX-Input-esp_dmx

someweisguy, 2023-04-08:
I've just submitted #61 as a draft.

It implements responding to RDM discovery packets only so far. It does not yet support anything useful. It implements responding by calling some code in `dmx_receive()` to parse packets and send a response. Therefore, if you do not call `dmx_receive()` your device will not send an RDM response. 

As a draft, it is a work in progress which completely lacks polish. But it should now be easier to see progress or provide input!

netmindz, 2023-04-08:
Great to see some progress on this. Are you at the state yet that you can respond to discovery with the fixture capabilities?

bethlenke, 2023-04-08:
I sat down with my EE friends and wrote some stuff based on arneboe's solution!

we implemented a way to:
set your on man UIDS
set the software version
set the device model description
set the manufacture label
create sensors to read of all the stuff supported by rdm (we used temp)
use identification
away to easily switch personality's (for me at least)

if  interested look at my fork!
https://github.com/bethlenke/esp_dmx

also here is the project's code to help if needed (warning, it is ugly).
[sketch_mar15a.txt](https://github.com/someweisguy/esp_dmx/files/11183086/sketch_mar15a.txt)

someweisguy, 2023-04-08:
> Are you at the state yet that you can respond to discovery with the fixture capabilities?

Not quite yet. My plan is to first implement the PIDs listed in the readme [here](https://github.com/someweisguy/esp_dmx#currently-supported-rdm-pids). A lot of the code that I wrote in the RDM controller is reusable for this function, so hopefully I can get a rough draft out this weekend. I will update #61 as I get those PIDs working. 

> if interested look at my fork!

Thank you! I will take a look - this is super helpful!

closed 2023-06-23 by someweisguy