someweisguy/esp_dmx #17: Won't compile for esp32-C3

txbugeater, 2022-04-22:
Attempted to build for the esp32-C3 and it fails with a compiler error as was suspected in the FIXME: comment on line 17 in the driver.h file.  The eps32-C3 only has two UARTS which is why it fails. I adjusted the #define but it still won't compile due to a different uart_struct.h file (in the esp-idf/components/esp32c3/include/soc dir) being used. 

I attempted to make appropriate changes to the data structure members and was able to get it to compile.  However, I have not yet succeeded in getting it to work. It does run and install the driver, but no activity on the io pins (yet). Most of the data structure member differences had similar equivalent member names between the two different uart_struct.h files. However, there did not seem to be an equivalent  member for tick_ref_always_on.  I made a wild guess on this one and I plan to revisit my changes in the areas where I changed tick_ref_always_on to another member (mem_clk_en).

someweisguy, 2022-04-22:
Good find. Thanks for opening this issue! You already know this, but for those who may find this issue without having seen the post on Reddit: I'm out of town this weekend and a bit busy the next, but I'll see if I can find time to investigate!

txbugeater, 2022-04-25:
While I am getting transitions on the TX line there is clearly still a problem. It does not send a valid packet yet.  There is also an issue with running as well.  It does not always stay running and it triggers the watchdog due to too tight of a loop somewhere. About 1 out of 5 or so resets of the processor it will not trigger the watchdog and I get continuous traffic on the scope.

txbugeater, 2022-04-28:
Made further progress and through some hacking, I believe I have the driver transmitting proper DMX packets.  I needed to change the clock to use when setting the baud rate.  I ended up using the RTC_CLK_FREQ (20MHz).  This produces the same frequency (22.73kHz) when measuring the TX signal.  It now matches the ESP32 based board.  I also observed that breaks were not being generated by the UART.  I have not investigated as to why it is not yet, but as a simple test of this being the problem with sending proper DMX packets, I sent a break in software after a new packet of data was memcpy'd in the interrupt routine (inter_handler.h in the block near line 66 ).

Next step is to test the receive side of the driver.

someweisguy, 2022-05-05:
Finally got around to looking at this - thank you so, so much for your patience. 

Everything you've said so far makes sense. I've updated the #defines in driver.h [here](https://github.com/someweisguy/esp_dmx/blob/e5c02634906317190af41ad9cdac916c91d49ee5/src/dmx/driver.h#L13) and like you mentioned that seems to make that file happy. 

For the issues in hal.h, the code is basically copied verbatim from appropriate files found [here](https://github.com/espressif/esp-idf/blob/v4.4.1/components/hal/esp32/include/hal/uart_ll.h). The folders for other ESP32-like devices are [here](https://github.com/espressif/esp-idf/tree/v4.4.1/components/hal). It sounds like you're already figured this out as well. It looks like the esp32-c3 defines a struct of unique types for each register in the UART, which is annoying. But as long as the UART hardware isn't too drastically different it should be okay to just swap out code for the specific device.

Which brings us to `uart_signal_conn_t`. I have no idea what these do, except that I know funky things happen if they're not configured correctly. Again, I copied these from the IDF. The ESP32-C3 version of `uart_signal_conn_t` looks very different than the ESP32 version. It looks like the values you've written in yours agrees with what I'd expect them to be. 

I've got a ESP32-C3 coming in the mail, so I am looking forward to hooking mine up to a scope and seeing what happens.

txbugeater, 2022-05-05:
I will share my changes (just hacks and not something you would consider checking in) when I get back from my vacation around the 15th if you need them.  You might have it all working by the time I get back!

I did test the receiver path and it seems to work with any further changes.

someweisguy, 2022-05-14:
I've been looking at a lot of code, but not writing a lot of code. But I've seen some changes in the ESP-IDF that look promising. 

When I released esp_dmx v1.1, the first release to support Arduino, the Arduino-ESP32 was based on ESP-IDF v3. Now, the latest version of Arduino-ESP32 is based on ESP-IDF v4.4.1. It looks like ESP-IDF v4 lets you include lots of code in the UART driver code that v3 obfuscated. Because esp_dmx creates its own, custom UART driver having those things be un-obfuscated is very handy.

So I think that there are enough changes here to be a major update. Since a lot of the esp_dmx code is copy/pasted from the ESP-IDF (due to the code obfuscation in ESP-IDF v3), I am going through and seeing what we can delete from the library and instead include straight from ESP-IDF. I'll be pushing these changes to the `dev` branch. I've also made a `dev-c3` branch as well.

someweisguy, 2022-05-15:
Well, I kinda-sorta made some progress in the `dev` branch. 

I changed how cmake builds the library. It now includes all files in `src/include` for end-users to include if they would like. Everything in `src/driver` is privately included so it can be used within the library, but the user cannot include these files. Then I have a folder called `src/esp32` which contains HAL functions for the ESP32. If you want to target the ESP32-C3, you would need to make an identical folder called `src/esp32c3` with the appropriate HAL functions. 

It looks like this works pretty well, except for in the Arduino IDE. The Arduino IDE has a different build process, which I need to read about and figure out how it works.

someweisguy, 2022-05-15:
More progress in the `dev` branch!

Like I mentioned earlier, one of the handy things about ESP-IDF v4.4 is that there are a lot of new exposed functions that weren't exposed before. So instead of using CMake to conditionally build the library which, to my knowledge, isn't supported in the Arduino IDE, I'm using those newly exposed functions. 

These functions are in the [UART HAL](https://github.com/espressif/esp-idf/blob/v4.4.1/components/hal/include/hal/uart_hal.h). These functions aren't public API and are subject to change at any point. But these functions are conditionally built depending on your hardware target in the Arduino IDE. I've wrapped the ESP-IDF UART HAL functions in this library's HAL function to make updates easier when the IDF HAL functions inevitably change.  

There are only 4 HAL functions that this library needs that don't exist in the ESP-IDF. I'm thinking it should be trivial to implement those per hardware target using preprocessor defines and ifdefs.

The `dev` branch is compiling for ESP32-C3. Writing almost works, but it looks like for some reason it's only writing a frame of DMX every 1.2 seconds instead of continuously.

someweisguy, 2022-05-16:
The timing tool still won't work, but both reading and writing work on the `dev` branch. Give it a shot when you are able. :)

txbugeater, 2022-05-16:
Great! I will try to get to it this week. I just returned from vacation on Saturday.

Dan

Sent with [ProtonMail](https://protonmail.com/) secure email.

txbugeater, 2022-05-21:
I finally got around to trying the dev version and I am happy to report that it works as advertised!  No issues with compiling or running.  I ran my simple transmit and receiver test without error. One observation is that the pulse frequency is a lower now.  I am measuring 19.23kHz on the scope.  My DMX receiver still accepts it and is able to decode the values.

someweisguy, 2022-05-21:
Woohoo! Great to hear! 

The UART hardware on the ESP32 and the ESP32-C3 are a bit different; on the ESP32 you can select one of two clock sources for the UART but on the C3 you can select one of three clock sources. I am guessing that is where the low pulse frequency is coming from. I'll see what we can do about that.

someweisguy, 2022-05-21:
I found the reason for the discrepancy between the ESP32 and ESP32-C3 UART speeds. If you read section 24.4.5.2 in the [ESP32-C3 Technical Reference Manual](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf), you'll see that setting the UART mode to `UART_MODE_RS845_HALF_DUPLEX` sets the bits for `UART_DL0_EN` and `UART_DL1_EN` which causes the UART to delay for two bits after each byte is transmitted.

Wow - I've learned something very interesting about the UART hardware in the ESP32. When you set your UART to transmit two stop bits on the ESP32, the following code gets called:
https://github.com/someweisguy/esp_dmx/blob/fdd9082fa7b535dbbc89a1626cd2546435b606fc/src/dmx/hal.h#L201-L203

It seems that the original ESP32 UART hardware was not designed to transmit more than one and a half stop bits. But by using this hardware workaround to delay each byte by an additional bit-length, it APPEARS that the UART is sending an additional stop bit. Fascinating design flaw in the original ESP32.

In any case, in the [DMX Standard](https://tsp.esta.org/tsp/documents/docs/ANSI-ESTA_E1-11_2008R2018.pdf) this inter-byte time is called the `“MARK” Time Between slots`. According to the standard, its minimum length is 0 microseconds and its maximum length is <1 second. So as you mentioned, even with the extra 2 bits between each byte, it is still valid DMX. Despite this fact, I'm adding a fix to `dmx_hal_init()` that will remove the inter-byte time on the ESP32-C3.

someweisguy, 2022-05-21:
I haven't done thorough testing, but every library feature should now be supported on ESP32-C3. Let me know if you find something that still needs attention. Otherwise, once I publish the next release I'll close this. 

Thanks!

txbugeater, 2022-05-22:
Great news!  Glad you found the issue so quickly. I have run into quite a few issues and quirks on chips over the years. Finding workarounds is what I was tasked to do for many years at my previous job/company. You don't always find work-arounds for hardware issues, but amazingly it happens more often than you would think.  I will download it and check it out.

txbugeater, 2022-05-22:
The transmitter is now producing the 22.73kHz frequency that I was expecting!  Thank you for getting the driver to work with the esp32-3C!

someweisguy, 2022-06-07:
v2.0 is released and addresses these issues. Thanks so much!

closed 2022-06-07 by someweisguy