someweisguy/esp_dmx #18: dmx_port when not using SparkFun shield

erasmudj, 2022-05-23:
I am trying to figure out what to do with the dmx_port_t dmxPort = 2; section when I am not using a Sparkfun Shield but a RS-485 MODULE. 
If I remove it I get errors.

someweisguy, 2022-05-23:
Hopefully I can help! Take a look at the `dev` branch readme. I just recently added a section on how to wire a RS-485 module [here](https://github.com/someweisguy/esp_dmx/tree/dev#wiring-an-rs-485-circuit). 

`dmx_port_t` is a type that tells your ESP32 which UART port to use for DMX. Depending on which ESP32 you have, you can have up to 3 UART ports. If you're familiar with `Serial` in Arduino, it's like choosing `Serial`, `Serial1`, or `Serial2`.

erasmudj, 2022-05-24:
Thank you. Now I understand.

closed 2022-05-24 by erasmudj