someweisguy/esp_dmx #19: Questions about Mark Before Break (MBB)

ok5884, 2022-05-25:
Thank you for the wonderful library.
Is it possible to control the Mark Before Break (MBB) 0S-1S High level (between frames) time in DMX transmission mode of this library?
I want to add Mark Before Break of 〇〇〇usec.

someweisguy, 2022-05-25:
Thanks for the kind words!

The short answer is no. It's not currently possible to control MBB times with this library.

The long answer is also no, but I'll explain how the ESP32 hardware works a bit. The ESP32 UART hardware allows you to send data like normal. But what it does differently than some other UARTs is that it has hardware that allows you to send a break and mark-after-break AFTER you send your UART data. As you know, DMX specifies that the break and MAB gets sent BEFORE the data. Many other DMX libraries perform busy-waits in order to send the break and MAB. This works fine, but it means that the break and MAB time are often imprecise (because busy-waiting is difficult to do with microsecond precision) and inefficient (because busy-waiting means that you can't do any work while you're busy-waiting). This library performs 1 busy-wait to send the initial break and MAB, but then after that it uses the hardware, post-data break and MAB to make it look like the break and MAB is being sent before the data. 

A feature that I've been putting off for a while is to use one of the two hardware timers on the ESP32 to send the break and MAB before the data. Using the hardware timers, it would be possible to precisely set the times for break, MAB, MBB, and even inter-byte time. I'd be happy to take a look into making this happen, but I can't promise it would happen soon. 

Anyway, I hope this helps!

ok5884, 2022-05-26:
I understand MBB. Thank you for your kind explanation. I learned a lot.
I have only touched Arudino, but I will also touch ESP-IDF to understand ESP.
I have high expectations for the introduction of hardware timers in the future.

someweisguy, 2022-06-24:
This feature will be available in the next release, so I am going to close it for now. Thanks for your patience!

closed 2022-06-24 by someweisguy