someweisguy/esp_dmx #20: Error on compiling

erasmudj, 2022-06-08:
I get the following error:
invalid conversion from 'int' to 'gpio_num_t' [-fpermissive]
on the line of:
dmx_set_pin(dmxPort, transmitPin, receivePin, enablePin);:

when I try to compile the DMXWrite example.

nitrokart, 2022-06-08:
gpio_num_t = GPIO number. In e.g. GPIO16, gpio_num should be GPIO_NUM_16

If you use the example:
transmitPin = 17;receivePin = 16;enablePin = 21;
dmx_set_pin(dmxPort, transmitPin, receivePin, enablePin);
These compiles fine:
dmx_set_pin(dmxPort, GPIO_NUM_17, GPIO_NUM_16, GPIO_NUM_21);

someweisguy, 2022-06-08:
Oof. This is a big dummy error on my part and I apologize!

The solution that @nitrokart proposed should work. I'll updated the examples!

someweisguy, 2022-06-08:
This has been fixed with v2.0.1! If you are using the Arduino Library Manager, it may take a few hours for v2.0.1 to appear. If you're using PlatformIO, the new version should be available for download momentarily. Or you can clone v2.0.1 directly into your library folder. 

Let me know if I can assist further and thanks so much for your patience.

closed 2022-06-08 by someweisguy