someweisguy/esp_dmx #23: Compilation problem in platformIO

arpschuino, 2022-07-30:
Hi, your library look fine !
I am curently testing it, no problem in the arduino IDE but in platformIO I have this error messages : 
.pio/libdeps/esp32doit-devkit-v1/esp_dmx/src/esp_dmx.h:12:28: fatal error: hal/gpio_types.h: No such file or directory
.pio/libdeps/esp32doit-devkit-v1/esp_dmx/src/dmx_types.h:19:3: error: unknown type name 'uart_sclk_t'

someweisguy, 2022-07-30:
Thanks for the kind words!

Which version of esp_dmx are you using? And are you using Arduino on PlatformIO?

arpschuino, 2022-07-30:
Thank you for this quick response!
I am using ESP_DMX version 2.0.1. with Arduino on PlatformIO.
If you want to get an idea of what I'm working on, here is my site: http://arpschuino.fr/index_e.php .
I am switching from atmega to ESP32.

someweisguy, 2022-07-30:
Fortunately this is an easy fix! The problem is that PlatformIO uses an out-of-date version of the Arduino-ESP32 framework. I thought that this would have been fixed by them since the new version was released nearly three months ago.

To fix the issue, you will have to update the platform in your `platformio.ini` file. Change the line that says `platform = espressif32` so that it reads `platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream`. Then you can rebuild your project. 

I'm hoping that esp_dmx v3.0 will fix this problem for good. Let me know if you run into any issues in the meantime.

Your project looks very cool! I will definitely take some time to look over your site. :)

arpschuino, 2022-07-30:
I tried including in c_cpp_properties.json > includePath:
"C:/Users/utilisateur/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/hal/include"
Without success...

someweisguy, 2022-07-30:
Updating include paths shouldn't be needed. You must inform PlatformIO to use an updated platform. This can be done in the `platformio.ini` file which can be found in the root of your project directory. 

Right now, your `platformio.ini` file should read something like this:
```
[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
lib_deps = someweisguy/esp_dmx@^2.0.1
```

If you update the `platform` to `platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream` your project should be able to be compiled.

arpschuino, 2022-07-30:
Thank you !
Well i tried that change in platformio.ini. 
I now have cryptic error messages :
************************************************************************************************************************************************************************

 #include <stddef.h>
          ^~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\libeea\AsyncTCP\AsyncTCP.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\src\Port.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\src\arp_server.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\src\_wifi.cpp.o] Error 1
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\src\Custom_port.cpp.o] Error 1
In file included from C:/Users/utilisateur/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.h:24,
                 from C:/Users/utilisateur/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.cpp:21:
c:\users\utilisateur\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\memory:62:10: fatal error: bits/stl_algobase.h: No such file or directory
 #include <bits/stl_algobase.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib9a0\FS\FS.cpp.o] Error 1

someweisguy, 2022-07-30:
Hmm... Well to start, it appears that your project is unable to find the file `bits/stl_algobase.h`. There also appears to be an issue with `stddef.h` but I am not sure what the problem is exactly. I know that esp_dmx does not directly include either of those files. Therefore, I am guessing that the problem is something to do with how your project is configured.

I am not sure I can help with this problem very efficiently. I would try to remove your code except for the basic `void setup()` and `void loop()` to see if your project compiles. Then start adding code segments back in to see what is causing this issue.

arpschuino, 2022-07-30:
I will try, thanks for your help.
It's a very big code, I started more than a year ago with version 3.5.0 of espressif32.
my code does not compile from version 4.0.0.
I'll let you know

someweisguy, 2022-08-01:
Closing this since it appears the issue has been resolved. Let me know if I can help with anything else! :)

closed 2022-08-01 by someweisguy