someweisguy/esp_dmx #24: very slow execution of the rest of the code

arpschuino, 2022-08-01:
Hi, sorry to disturb you again !
is it possible to bypass this verification :
if (xQueueReceive(queue, &packet, DMX_PACKET_TIMEOUT_TICK)
"The macro DMX_PACKET_TIMEOUT_TICK can be used to block the task until a packet is received or a DMX timeout occurs." So I don't want to block the execution of the rest of my code !

arpschuino, 2022-08-01:
In summary, this is what my main loop consists of:
If you receive DMX:
read it.
else if you receive artnet wifi:
read it.
perform other actions such as running stepper motors

thank you for your attention

someweisguy, 2022-08-01:
Yes, this is possible. The macro `DMX_PACKET_TIMEOUT_TICK` specifies how long to block code execution. If I recall correctly, it is about 1.25 seconds. 

If you replace the macro with zero (in other words, write `xQueueReceive(queue, &packet, 0)` the program will not block.

arpschuino, 2022-08-01:
OK perfect, thank you.
So if I understand correctly, it is better to set this value to Zero rather than not doing this check. And I will still be able to receive DMX correctly?

someweisguy, 2022-08-01:
In most cases, using `xQueueReceive()` is preferred. 

As long as the mode is set to `DMX_MODE_READ` you will be able to receive data. Using `xQueueReceive()` allows you to get notified when a new packet has been received. This may be easier to explain with some example code:
```
// Example Code 1
uint8_t data[513] = {};
while (true) {
  if (xQueueReceive(queue, &packet, 0)) {
    dmx_read_packet(dmx_num, data, sizeof(data));

    // Process data...
    // Perhaps it takes a long time to process data!
  }
}
```
In this example, I process data every time I receive a new DMX packet. It takes a long time to process data, so I only want to process data when I receive a new packet. 

And here's a different example, example 2:
```
// Example Code 2
uint8_t data[513] = {};
while (true) {
  dmx_read_packet(dmx_num, data, sizeof(data));

  // Process data...
}
```
In this example, I read data in every loop. This is acceptable too! The difference here is that I'm processing data even if I haven't received a new packet. This can be considered inefficient to some. 

Hopefully this helps! Let me know if you have more questions! :)

arpschuino, 2022-08-02:
Thank you very much for this very precise and detailed answer, it is a pleasure to deal with people like you.

someweisguy, 2022-08-02:
Glad I could help! I'll close this for now. Feel free to reopen or open a new issue if you have any more questions!

closed 2022-08-02 by someweisguy