someweisguy/esp_dmx #25: Incompatibility with LittleFS

bensuffolk, 2022-08-16:
I find that if I try to write a file with LitteFS whilst received DMX I get a Guru Meditation Error.

The backtrace is corrupted so I can't fully see exactly where this is happening :

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x400e692c  PS      : 0x00060035  A0      : 0x40086abc  A1      : 0x3ffbf22c  
A2      : 0x00000078  A3      : 0x3ffbdcc0  A4      : 0x00000020  A5      : 0x3ffbdcbc  
A6      : 0x3ffbc2f8  A7      : 0x00000001  A8      : 0x8008165c  A9      : 0x00000078  
A10     : 0x3ffbdcbc  A11     : 0x3ffbc45c  A12     : 0x3ffbf244  A13     : 0x3ffbdcbc  
A14     : 0x3ffc4608  A15     : 0x84022044  SAR     : 0x00000017  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400865a9  LEND    : 0x400865b1  LCOUNT  : 0x00000027  

Backtrace:0x400e6929:0x3ffbf22c |<-CORRUPTED
  #0  0x400e6929:0x3ffbf22c in uart_ll_write_txfifo at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32/include/hal/uart_ll.h:235
      (inlined by) uart_hal_write_txfifo at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/uart_hal_iram.c:40

ELF file SHA256: 0000000000000000

Rebooting...
```

I am using a ESP WROVER-E and have the DMX being received on UART1 (But I have also tried it on UART2 and there is no difference).

I'm using the Arduino Framework which reports to be using ESP-IDF v4.4.1-1-gb8050b365e

bensuffolk, 2022-08-16:
I have taken the example dmx read and added a couple of lines to it (I have also taken out all the comments to keep it brief here, and added a couple of comments to show where the LitteFS code was added). This will cause the Panic every time in a continuous loop:

```

#include <Arduino.h>
#include <esp_dmx.h>
#include "FS.h"
#include <LittleFS.h>

int transmitPin = 5;
int receivePin = 19;
int enablePin = 18;

dmx_port_t dmxPort = 1;

byte data[DMX_MAX_PACKET_SIZE];

QueueHandle_t queue;

unsigned int timer = 0;
bool dmxIsConnected = false;

void setup() {
  Serial.begin(115200);

  /*
  * Open the file system
  */
  if(!LittleFS.begin(true)) {
   Serial.println("Formatted");
   LittleFS.mkdir("/dmx");
  }

  dmx_config_t dmxConfig = DMX_DEFAULT_CONFIG;
  dmx_param_config(dmxPort, &dmxConfig);
  dmx_set_pin(dmxPort, transmitPin, receivePin, enablePin);

  int queueSize = 1;
  int interruptPriority = 1;
  dmx_driver_install(dmxPort, DMX_MAX_PACKET_SIZE, queueSize, &queue,
                     interruptPriority);
}

void loop() {
  dmx_event_t packet;

  if (xQueueReceive(queue, &packet, DMX_PACKET_TIMEOUT_TICK)) {

    if (packet.status == DMX_OK) {

      if (!dmxIsConnected) {
        Serial.println("DMX connected!");
        dmxIsConnected = true;
      }

      dmx_read_packet(dmxPort, data, packet.size);
      timer += packet.duration;

      if (timer >= 1000000) {
        Serial.printf("Start code is 0x%02X and slot 1 is 0x%02X\n",
                      data[0], data[1]);
        timer -= 1000000;
      }

    } else {
      Serial.println("DMX error!");
    }

   /*
    * Save the DMX Packet to a file
    */
   File file = LittleFS.open("/dmx/00.dmx", "w");

   for(uint16_t x=0; x<DMX_MAX_PACKET_SIZE; x++) {
    file.write(data[x]);
   }
   
   file.close();

  } else if (dmxIsConnected) {
    Serial.println("DMX timed out! Uninstalling DMX driver...");
    dmx_driver_delete(dmxPort);

    while (true) yield();
  }
}

```

someweisguy, 2022-08-17:
Thanks so much for the detailed report and the example code. I am hoping this is a simple fix. 

Because DMX is a continuous byte-stream of data, when writing this library I opted to put the DMX related interrupts into the ESP32's Instruction RAM, which would allow for low latency operation. However, because a good portion of the DMX code is in IRAM, it means that the ESP32 will crash when attempting to access cached data. So hopefully, the solution is to stop the library from attempting to access cached data. I will let you know what I find!

someweisguy, 2022-08-17:
I am able to reproduce the error. It appears the error is related to the pins being used for DMX. Are you able to use different hardware pins? I found that the following pins fixed the crash.
```
int transmitPin = 17;
int receivePin = 16;
int enablePin = 21;
```
I am not 100% positive, but I believe LittleFS uses SPI and because hardware pins 18, 19, and 5 are the default SPI pins, this may be what is causing the crash.

bensuffolk, 2022-08-17:
Thanks for looking so quickly.

This is my first real ESP32 project, and one of the things I really liked about the ESP32 was the GPIO matrix and being cable to move IO GPIO around. I knew there were some pins you should not use as it causes boot problems, and some IO GPIO connected to the internal SPI flash.

[https://randomnerdtutorials.com/esp32-pinout-reference-gpios/](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/) seems to suggest the internal SPI flash are on GPIO 6 through 11, which I have avoided (and had assumed would be used for SPI involved in LitteFS for the internal Flash access)

Moving the pins is not ideal as I have a lot of pins in use and prototyped a PCB based on being able to use these pins. 

I have an SPI ethernet (W5500) connected using the second SPI bus on GPIO 15,14,13,12 and using 4 for interrupt and 2 for reset.

I have I2C running on 25 & 32

Then 2 UARTS for DMX (one to send and one to receive) send on TX = 21, RX = 23, EN = 22. Receive on TX = 5, RX = 10, EN = 18

Then 3 LED outputs on 27, 26, 33

I designed the PCB with the full RX / TX / EN lines so I could support RDM over both ports. But I guess in the short term I could potentially just look to use a single RX GPIO for DMX In, and a single TX pin for DMX Out, which means I only need 2 pins instead of 6, but I don't really want to do PCB redesign if I can solve this in software.

Looking at the WROVER datasheet GPIO16 & GPIO17 are both connected to the internal PSRAM so are not exposed as pins. (I'm also using PSRAM in my application, which is why I opted for the WROVER as opposed to the WROOM).

I will do a little more research and see what I can find. Thanks for your help so far.

bensuffolk, 2022-08-17:
A quick update, I have tried the second set of pins (21,23,22) as they are connected to a driver circuit anyway (as it was to be used for sending) but I get the same results. Maybe I have been unlucky in the pins I picked.

I have posted a query on the ESP32 forum (still waiting to be approved by a mod) so we will see if I get any extra infor / help from there. 

In the mean time I have removed the IRAM flag from the interrupt whilst I develop the rest of the application. Since I am using LittleFS only occasionallyI might have to live with the possibility of a buffer overflow if no other solution can be found.

For a bit of context, I'm developing a DMX playback device. So only very occasionally when a user wants to save a snapshot of the universe will I have to use LitteFS. I can load all the saves snapshots into PSRAM before starting the DMX handlers.

The only other situation for FLASH access will be firmware upgrades, which I can disable the DMX for before that happens.  So I think I can control whats going on with FLASH pretty well.

Maybe a different option would be to disable the interrupts when I'm going to use the FLASH, then they can be active the rest of the time in IRAM to make sure there are no issues?

someweisguy, 2022-08-17:
I'm recategorizing this as a "wontfix" because this is a general issue with ESP32 - figuring out which pins you can use and can't use are a pain.

I think you are on the right track - I was going to suggest the possibility of disabling interrupts when you do flash read/write operations. I am not sure if that will break LittleFS, but it is worth a try. 

Another option I was going to investigate is to detach your SPI pins from 5, 18, and 19. Again, I am not sure if this will break LittleFS.

someweisguy, 2022-08-17:
Wait a minute - I think I may see what is happening here. This may be a bug after all!

If I understand correctly, your panic is indicating that the ESP32 is attempting to call the function `uart_ll_write_txfifo()` from memory address `0x400e6929`. This memory address is in cache, which makes sense because you're getting the cache disabled crash. 

That function should not be in cache, it should be in IRAM. I think what may be happening is that Arduino tells the ESP32 not to place UART functions in IRAM. Since UART functions are not put in IRAM, they are put in cache, and it is causing your ESP32 to crash. 

I'm a bit busy today, but I will do some experimentation this week and get back to you!

bensuffolk, 2022-08-17:
That sounds promising, thank you!

someweisguy, 2022-08-19:
Thanks for the excellent find! This has been fixed in v2.0.2. The issue was indeed that certain functions were not being put in the correct memory addresses. Instead, they were being put in cache, which was causing problems when cache was disabled by LittleFS.

If you're using PlatformIO, the updates should be available to download shortly. It may take a few hours if you're using the Arduino Library manager. Or you can download the v2.0.2 source files directly from GitHub.

closed 2022-08-19 by someweisguy

bensuffolk, 2022-08-20:
Thanks for fixing it so quickly.

I have been using code I wrote over 10 years ago for a pic micro in ASM for all my DMX stuff, and I was thinking I would have to write it all over again moving to ESP, but you have do a great job. Thank you so much.