someweisguy/esp_dmx #27: Send basic data

CristianRM7, 2022-08-26:
Hi, the library is amazing, great job! How can I send the value 255 through channel 2? Thank you!

someweisguy, 2022-08-26:
Thanks for the kind words!

To write a value to a specific DMX address, you can use the `dmx_write slot()` function. Example code would look something like this:

```
dmx_write_slot(dmx_num, 2, 255);  // Write the value 255 to address 2
dmx_send_packet(dmx_num, DMX_MAX_PACKET_SIZE);  // Send the data to any DMX devices
```

Hope this helps! Let me know if you have any other questions

CristianRM7, 2022-08-29:
Thanks for answering! I think I'm missing something, I have a dmx spotlight that I can program in different ports (from 1 to 512), I have it in 1 for example. And then to put it in red, I have to send to that port 1, CH2 the value 255, which in other libraries what I send is for example: DmxSimple.write(2,255);
Here I have:
dmx_port_t dmxPort = 1;
dmx_write_slot(dmxPort, 2, 255); // Write the value 255 to address 2
dmx_send_packet(dmxPort, DMX_MAX_PACKET_SIZE);

But it sends me all the RGB values. Am I missing something? Thank you!

someweisguy, 2022-08-29:
Hmmmm. You would also need to install the DMX driver and set the driver to writing mode. Do you mind sharing your code? Or have you tried modifying the [DMXWrite.ino](https://github.com/someweisguy/esp_dmx/blob/release/v2.0/examples/DMXWrite/DMXWrite.ino) example?

CristianRM7, 2022-08-30:
Of coruse!! 
`#include <esp_dmx.h>

int transmitPin = 17;
int receivePin = 16;
int enablePin = 21;

dmx_port_t dmxPort = 1;

void setup() {
  Serial.begin(115200);

  dmx_config_t dmxConfig = DMX_DEFAULT_CONFIG;
  dmx_param_config(dmxPort, &dmxConfig);
  dmx_set_pin(dmxPort, transmitPin, receivePin, enablePin);

  int queueSize = 0;
  int interruptPriority = 1;
  dmx_driver_install(dmxPort, DMX_MAX_PACKET_SIZE, queueSize, NULL, interruptPriority);

  dmx_set_mode(dmxPort, DMX_MODE_WRITE);
}

void loop() {
  dmx_write_slot(dmxPort, 2, 255);  // Write the value 255 to address 2
  dmx_send_packet(dmxPort, DMX_MAX_PACKET_SIZE);  // Send the data to any DMX devices
}`

someweisguy, 2022-08-31:
Thanks for the example code! I uploaded it to my ESP32 and it is running exactly how I would expect it to. I plugged the ESP32 into an oscilloscope and I confirmed that the results are what I expected. Address 2 is being sent the value 255 (hex 0xFF). Here's a screenshot if you're interested. 

[image]

I also ran some example code of DMXSimple and plugged it into an oscilloscope. I'm not sure I understand DMXSimple very well. I think I must have done something wrong because the output was not within DMX specification. I think that DMXSimple may start their array at 1 and esp_dmx starts at 0. So perhaps, instead of calling `dmx_write_slot(dmxPort, 2, 255)` you might try `dmx_write_slot(dmxPort, 1, 255)`?

I am so sorry that you are still having trouble, but hopefully this helps!

someweisguy, 2022-09-09:
Closing this due to lack of activity. Let me know if I can help with anything else!

closed 2022-09-09 by someweisguy