someweisguy/esp_dmx #28: two driver

nada0302, 2022-09-20:
Thanks for this library.

It works well with two drivers also (for TRANSMITTING and RECIEVING).
I just made below adjustment for initialization.
byte data[DMX_PACKET_SIZE] = {}; (add "= {}"; from reference code.)

By the way, is this two driver "unusual case" ? and better not to do as suggested on README ?
It is pretty nice if two driver or more can be used.

I read through README, I found below wording.
> If transmitting and receiving data simultaneously is desired, the user can install two drivers on two UART ports. 
> It should be noted that this is an unusual use case. This library is not meant to act as a DMX optoisolator or splitter.

someweisguy, 2022-09-20:
Thanks for the kind words! And it's good to hear from you again!

Your adjustment looks like a good idea. By adding the curly braces, you would be initializing the data array to all zeroes. Leaving out the curly braces would just declare the variable, but the values would be garbage data. 

I mentioned in the readme that using multiple drivers is unusual because I can't say for sure if it would cause problems and most users are probably only going to use the one driver. You are welcome to use 2 drivers at once if you aren't having any issues. I haven't thoroughly tested using multiple drivers at the same time, so I can't say for sure if it would cause bugs.

Hope this helps! Let me know if you have any other questions.

nada0302, 2022-09-21:
Thanks for your quick reply & remembering me. ^^)

Good to learn your comment and I'll keep test.
(I do recording DMX signal together with submitting signal to lighting apparatus)

If any issue I cannot solve, I will ask for your help again and open the issue.
Thanks!

closed 2022-09-21 by nada0302