someweisguy/esp_dmx #30: Supporting esp-idf 5.x+

stonegray, 2022-10-07:
The current version does not build on `esp-idf` v5.0 (and 5.1). Release v5.0 is out of beta and is currently the most recent supported version for all chips. 

I've made some small changes to release/2.0 to get it successfully building over at https://github.com/stonegray/esp_dmx/commit/e340115e76c1c9c575490157ba69f72156755b44 and assume many of these changes can be ported forward to the master version.  This isn't PR-ready as I don't think it will build on IDF v4.4, which is currently the default. 

I'm not sure I've implemented `sclk_freq` correctly, and if this can properly use the reference clock instead of the default. 

Espressif has a [migration guide for moving from v4.4 to v5](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/index.html) (possibly incomplete, I didn't see the changes to `uart_hal_set_baudrate` and friends there)

someweisguy, 2022-10-09:
Thanks for bringing this to my attention. I've been keeping an eye on ESP-IDF v5 for a while - I am aware that this library won't build on it yet, so I appreciate the help!

I've been working hard on esp_dmx branch v3.0. That branch builds on IDF v4.4, but it definitely won't build on IDF v5. My plan is to update esp_dmx v3 to work with IDF v5. I don't plan on updating esp_dmx v2 so that it builds on IDF v5 but you are welcome to take a look.

The tricky part is that many users only use Arduino and it will likely be some time before Arduino supports IDF v5. So I am planning on using preprocessor macros to allow esp_dmx v3 to work on either IDF v4 or v5.

At this point, esp_dmx v3 supports full DMX support - I am just working on RDM and documentation. You're welcome to take a look at that branch if you would like!

stonegray, 2022-10-12:
I'm planning to update to esp_dmx v3 shortly, for now my patched v2 is working well enough I can update the rest my project. 

Are there any special considerations for supporting Arduino beyond ensuring v4.4 compatibility?

someweisguy, 2022-10-16:
The only consideration I can think about for Arduino is that, as far as I am aware, CMake does not work when using the Arduino IDE. All conditional building must be done with preprocessor defines. 

I'll get started with v3 as well. The first big change I know needs to happen is updating the hardware timers. Should be pretty straightforward changes but they'll result in somewhat big organizational changes in the library.

someweisguy, 2022-10-29:
Update on this: I was able to add functions in [dmx_context.h](https://github.com/someweisguy/esp_dmx/blob/release/v3.0/src/private/dmx_context.h) so that there are function that control the hardware timer. The functions call the appropriate hardware timer API depending on the major version number of ESP-IDF that is used. So far the ESP-IDF v5 code are just stubs, but it should be pretty straightforward to implement. The ESP-IDF v4 code still works, though I need to do more in-depth testing. 

Unless I'm missing something, ESP-IDF v5 is still in beta. I haven't seen a release for v5 on the [ESP-IDF repository](https://github.com/espressif/esp-idf). Until I see an official release for v5, I'm going to put this issue on the back-burner so that I can churn out esp_dmx v3 as quickly as possible.

Let me know your thoughts and if you have any questions!

someweisguy, 2022-11-08:
Just as an update on this - I actually decided to remove the functions in `dmx_context.h` in favor of calling the timer API directly. I've found that helps improve performance a somewhat noticeable amount.

someweisguy, 2022-12-13:
ESP-IDF v5 compatibility is in progress now. :)

weblaunch, 2022-12-31:
I would also appreciate a v5 compatible library :-) I've spent some time with both this version and @stonegray's and can't get either to run without fatal errors.

It would also be helpful for DMX noobs like me if the docs helped to understand how the ports, packets and slots translate into sending RGBW values to a light for example

someweisguy, 2023-01-01:
Oh that is a great idea! I can't believe I didn't think about that in the docs. 

I'm still working on v5 compatibility in the pull request [here](https://github.com/someweisguy/esp_dmx/pull/36). I haven't done much work on it in a while because the new [GPTimer API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gptimer.html) is much less precise than the hardware timer API in the v4 library. It makes me want to pull my hair out a bit! :)

If you want to get started with ESP-IDF v5 right now, the [feature/esp-idf-v5 branch](https://github.com/someweisguy/esp_dmx/tree/feature/esp-idf-v5) should be stable enough for most situations. The only two problems that I've found are:
1. There are some situations where the RDM output is not to specification. Responding devices should still be able to interpret these packets, but I'm still working to correct these problems.
2. If you're reading or writing to flash memory you might experience crashes using the feature branch as-is.

someweisguy, 2023-01-02:
This is done. The latest commit on release/v3.0 supports ESP-IDF v5. I'll put a esp_dmx v3.0.1 release out soon. 

Keep in mind that Arduino still does not support ESP-IDF v5. If you compile code on Arduino, Arduino will continue to use ESP-IDF v4.4 or earlier.

closed 2023-01-02 by someweisguy