someweisguy/esp_dmx #31: dmx_receive returns too early and data size is wrong

arneboe, 2022-10-29:
I am experimenting with the v3.0 release branch and am seeing some strange behavior.

- The `dmx_receive` method returns too early. In my experiment it is returning at ~150hz (Max hz of dmx is ~44hz).
- The size that is reported by dmx_receive is random. I.e. after a re-boot it will return a different size. The size never changes.

From my understanding of the documentation dmx_receive should block until a new dmx frame is returned but somehow this is not the case.

Am I missing something? This feels like I am not initializing something correctly.
The received data is correct, i.e. when I move the fader on the dmx sender I can see the correct value in the buffer.

The code below produces the following output:

```
...
E (15660) main: 15.675053: hz: 154.513034, ret: 255, data: 7
...
```

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "esp_dmx.h"
#include "esp_log.h"
#include "esp_system.h"

#define TX_PIN 17 // the pin we are using to TX with
#define RX_PIN 16 // the pin we are using to RX with
#define EN_PIN 21 // the pin we are using to enable TX on the DMX transceiver

static const char *TAG = "main";
static uint8_t data[DMX_MAX_PACKET_SIZE] = {};

void dmxTask(void *unused)
{
    ESP_ERROR_CHECK(dmx_set_pin(DMX_NUM_2, TX_PIN, RX_PIN, EN_PIN));
    ESP_ERROR_CHECK(dmx_driver_install(DMX_NUM_2, DMX_DEFAULT_INTR_FLAGS));
    int count = 0;

    while (true)
    {
        dmx_event_t event;
        size_t ret = 0;
        ret = dmx_receive(DMX_NUM_2, &event, DMX_TIMEOUT_TICK);
        if (ret)
        {
            // Check that no errors occurred.
            if (event.err == ESP_OK)
            {
                count++;

                const double currentTimS = esp_timer_get_time() / 1000000.0;
                ESP_LOGE(TAG, "%f: hz: %f, ret: %d", currentTimS, (count / currentTimS), ret);
                dmx_read(DMX_NUM_2, data, event.size);
            }
            else
            {
                ESP_LOGE(TAG, "dmx error: %s", esp_err_to_name(event.err));
            }
        }
    }
}

void app_main()
{
    TaskHandle_t dmxTaskHandle = NULL;
    // TODO determine real stack usage and reduce later
    xTaskCreatePinnedToCore(dmxTask, "DMX_TASK", 10240, NULL, 2, &dmxTaskHandle, 1);
    if (!dmxTaskHandle)
    {
        ESP_LOGE(TAG, "Failed to create dmx task");
    }
}
```

arneboe, 2022-10-29:
I did some digging and it seems that this bug/behavior was introduced in commit 834c1316.

someweisguy, 2022-10-29:
Great catch! I'll investigate.

someweisguy, 2022-10-29:
I did some poking around, and it appears this problem is much older than 834c131. I'm very frustrated with myself for this!

First, there was an issue where the function `dmx_receive()` was waiting for a task notification for much too long due to being passed an inappropriate variable. This would cause occasional watchdog crashes. I've fixed this error.

The second error which I _think_ I've fixed has to do with calling `dmx_receive()` after you already have called `dmx_receive()` but before actually receiving new DMX data. The expected behavior is that the function blocks until a new packet has been received. The function was instead returning early to report that a new packet has been received when it hadn't. This is why we were seeing 159Hz. 

The third error may need a bit of background first. A standard DMX packet is 513 bytes long. According to the DMX specification, however, there is no minimum packet size. I've used lighting consoles that output 197 bytes per packet.

someweisguy, 2022-10-29:
I did some poking around, and it appears this problem is much older than 834c131. I'm very frustrated with myself for this!

First, there was an issue where the function `dmx_receive()` was waiting for a task notification for much too long due to being passed an inappropriate variable. This would cause occasional watchdog crashes. I've fixed this error.

The second error which I _think_ I've fixed has to do with calling `dmx_receive()` after you already have called `dmx_receive()` but before actually receiving new DMX data. The expected behavior is that the function blocks until a new packet has been received. The function was instead returning early to report that a new packet has been received when it hadn't. This is why we were seeing 159Hz.

The third error may need a bit of background first. A standard DMX packet is 513 bytes long. According to the DMX specification, however, there is no minimum packet size. I've used lighting consoles that output 197 bytes per packet instead of the typical 513. There is some logic in the DMX driver that is intended to allow for shorter DMX packets to be processed. However, it appears this logic is not working as intended. The result is that the first several packets received are reporting a much shorter packet size than the actual size received. Eventually the DMX driver reports the correct packet size. 

I'm going to merge this branch with the `dev` branch and do some work to correct these errors.

arneboe, 2022-10-30:
wow! Thanks for the quick reply :-) 
I will do some more digging as well and try how to understand how the dmx driver works.

arneboe, 2022-10-30:
A question that has nothing to do with this ticket, feel free to ignore it :D (I am still new to the whole esp32 world, coming from stm32 :D)

I am trying to understand the `dmx_uart_isr` and was wondering if there is a reason why you set `uart_ll_set_rxfifo_full_thr` to 1.
If I understand the code correctly this means that you will get an interrupt for every byte that the uart receives. That seems somewhat cpu-time consuming. Wouldn't it be possible to set the buffer size to 513 and ignore the `DMX_INTR_RX_ERR`?
If the buffer is big enough it should be possible to just listen to the `DMX_INTR_RX_ERR` and `DMX_INTR_RX_BREAK` interrupts to get all dmx data. This would free a massive amount of cpu-time.

someweisguy, 2022-10-30:
Great question! What you're describing would work too. This library already uses a large FIFO for sending data, just not receiving data. The ESP32 UART is unusual because it has that UART FIFO. Most other microcontroller UARTs just have a single byte FIFO, as far as I know. It would be possible to setup the ESP32 so that it can receive multiple bytes of data before the interrupt needs to fire.

If I recall correctly, the base ESP32 has somewhere around 1KB of FIFO space shared between 3 UARTS. Each UART can be configured to use different sizes of FIFO. By default, each UART's FIFO is 128 bytes. I think it is possible to configure a UART so that its FIFO is 513 bytes long. The problem with this is that ESP-IDF doesn't actually support this feature in software. The 128 byte FIFO is actually a [hardcoded integer literal](https://github.com/espressif/esp-idf/blob/master/components/hal/esp32/include/hal/uart_ll.h#L282-L284) in ESP-IDF. I am not sure why they've done this. Perhaps this feature doesn't actually work in the silicon? Furthermore, the [ESP32 technical reference manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) doesn't explain very well how to expand the UART FIFOs. The registers controlling the FIFOs can be seen on page 364, but page 338 mentions that expanding the FIFO sizes can also overwrite the FIFOs of other UARTs. So the whole thing just seems messy and difficult. 

It would be possible to set the RX FIFO full threshold to the full 128 bytes, though! As you know, you can set the full threshold to whatever number you want. The ESP-IDF UART implementation uses 120 by default. When you have 120 bytes or more in your FIFO, the interrupt fires and you can read your data. If you only receive 119 bytes the interrupt doesn't fire. What happens instead is the ESP32 waits until a user-specified timeout occurs. When that timeout occurs the interrupt fires and you read your data. I think I went with the 1 byte full threshold because I concerned about latency - particularly when it comes to using RDM. Now that I am thinking about it again though, I'm not sure that latency would be a problem. The minimum RX FIFO timeout length would be somewhere around 44 microseconds. This would mean that you receive 119 bytes and then 44 microseconds later you can read data. I might play with this and see what happens!

someweisguy, 2022-11-01:
This appears to be fixed. I've merged changes back into the `release/v3.0` branch. Let me know if you have any questions or if you find any more issues! :)

closed 2022-11-01 by someweisguy

arneboe, 2022-11-01:
Thanks for the detailed explanation.
The 128 byte limit seems a bit strange and smells like a workaround for a bug in silicone :)
I don't understand why you would need the timeout interrupt at all in the dmx use-case.
From my understanding in dmx you either get another byte or you get a break. The break (or the frame error) should be detected when the stop bits of the 514th byte are low, this happens ~44us after the last byte was received. 

And thanks again for the quick fix! I am back at work but will continue playing with the esp during the weekend :)