someweisguy/esp_dmx #34: adding ESP8266 support

Theexternaldisk, 2022-11-19:
Hi quick question what would it take to add ESP8266 support?
Would love to use this on a project to use an ESP8266 as DMX reciever

someweisguy, 2022-11-19:
Great question - I think it would take a lot of work. This library uses a lot of functions defined in the ESP-IDF which only supports ESP32. The [framework for ESP8266](https://github.com/espressif/ESP8266_RTOS_SDK/tree/v3.4) says that the ESP-IDF will support ESP8266 soon, but it said this 2 years ago and I haven't seen any progress to achieve this goal.

I don't think it's necessarily impossible. This library has been written to be somewhat portable. This library uses a HAL which could easily be updated to include code that only builds when ESP8266 is detected. And the ESP8266 UART hardware appears to be similar to that of the ESP32.

But right now I am working on implementing RDM in v3 of this library. It's a good amount of work and I wouldn't be surprised if it takes at least another six months until it's complete. Because of this, I wouldn't hold my breath on ESP8266 support. If you'd like to take a crack at it, you are welcome to fork this repo! Or you can try a library like [DMXSerial](https://github.com/mathertel/DMXSerial) if you're using Arduino.

Let me know if you have any questions! :)

someweisguy, 2022-11-19:
I'll add that the only things needed for v3 of this library to work are:

- FreeRTOS v8.2.0 or greater (for Task Notification API)
- A UART port with the ability to invert the TX signal
- A dedicated hardware timer per DMX port with microsecond resolution
- Edge-Triggered GPIO Interrupts (for the DMX sniffer)

I believe that the ESP8266 framework has all of these.

Versions 1 and 2 of this library require the use of specific features of the ESP32 UART hardware so I'd say that porting those branches to ESP8266 are impossible.

Theexternaldisk, 2022-11-19:
Thank you for your reply. Forking your repo is a bit above my head :) 
DMXSerial unfortunately is written specificly for AVR from what I can read from the library.
Keep up the good work!

closed 2022-11-23 by someweisguy