someweisguy/esp_dmx #37: Using esp_dmx for reading and writing at the same time

Pit1966, 2022-12-27:
I tried to set up device that patches DMX data. So i tried receiving and sending at the same time (with two different RS485 converters, one for rx and one for tx).
But it does not work very well in parallel. DMX read errors come up frequently.
I combined the read and write examples in some way.
But what is the  recommended method to receive data on one DMX input and send it (with changes) to another DMX output?

someweisguy, 2023-01-01:
Thanks for the question!

DMX is a unidirectional protocol so it doesn't support sending and receiving data at the same time. The exception is with using RDM, the extension to the DMX protocol. The best way to receive and send DMX from one ESP32 is to install 2 drivers; one to receive and one to send.

If this is what you are attempting, let me know and I can investigate further!

Pit1966, 2023-01-03:
Am Sonntag, 1. Januar 2023, 01:14:02 CET schrieb Mitch Weisbrod:
> Thanks for the question!
>
> DMX is a unidirectional protocol so it doesn't support sending and receiving data at the same time. The exception is with using RDM, the extension to the DMX protocol. The [...]

Thank you. Yes, I know DMX is unidirectional. But I am trying to receive and transmit on two different lines. The device I am building is a kind of PC DMX interface plus merger function.
I tried with 2 drivers. But this did not work very well. I got flickering on output line or dropouts on input.
And a drawback is, that I have to use 2 uarts of the esp32.
My goal is to have 2 input lines plus 2 output lines. (double DMX universe merging) and additionally the "normal" serial interface connected to a PC.
I have managed to implement a completely interrupt driven software where a single UART handles the rx and tx part at the same time.
If you like I can post the code.

someweisguy, 2023-01-04:
Thanks for the info. A few more questions:

- Which version of esp_dmx are you using?
- Are you using ESP-IDF or Arduino?
- If you are not  using Arduino, I am assuming you are using ESP-IDF v4.4. If you're using ESP-IDF v5 let me know!

If you could post some code, that would be helpful. I will try to recreate this problem this evening!

someweisguy, 2023-01-05:
I wrote a sketch which receives one universe of DMX and transmits that same universe out on a different driver [here](https://gist.github.com/someweisguy/cbcbe4558ea0b77e3cbedddf4ce52579). I found a bug in the `dmx_receive()` function when you pass to it a block time of 0. It causes this code to work not quite as I expect; the output DMX packet-rate decreases when there is no input DMX.

Upon rereading your last comment I think I may have been misunderstanding what you are trying to do. It sounds like you want to handle a total of four universes with 2 UART/DMX lines. I've never seen an existing piece of hardware do something like this so I'm still a bit confused about what your goal is. Thank you for your patience while I figure out how to help!

So if you're building a DMX interface with a merger function, you want to be able to connect the ESP32 to a PC so that your PC can send commands to the ESP32 to output DMX. Then the remaining 2 inputs and 1 output will be able to merge 2 universes into 1 output universe. Is this correct?

Pit1966, 2023-01-07:
First of all, thank you for investigating that and also for providing the source code.
Yes, maybe we have been misunderstanding us, but your conclusion is correct.
My goal is to build a device with 2 input dmx universes and 2 output dmx universes. These universes should be managed by PC commands. Later maybe ArtNet support will be added.
Funny, I wrote a 2 driver test sketch, that looks very much like yours. You find it [here](https://gist.github.com/Pit1966/a8f3d550201666addbef6aaca3599938).
If you are interested, I wrote a sketch that does DMX forwarding like your test sketch, that is completely interrupt driven. For now it uses only 1 input and 1 output universe, but I think this can be expanded.
Here is the [DMXbox sketch](https://gist.github.com/Pit1966/6774f6716c5478d73b3519bcad5525a2).
By thy way, I am using Arduino framework, but my IDE is sloeber.

someweisguy, 2023-01-07:
Thank you for the additional information! I have some more investigation to do on this, but I'll let you know what I've found so far. 

What I know is that this library only allows either sending or receiving at a given time. This is because of the RDM compatibility I added recently, though I had the same philosophy in earlier versions of the library as well. In other words, in versions 1, 2, and 3, you can only ever _either_ send _or_ receive DMX data with a DMX driver at any given time.

When calling the function `dmx_send()`, the UART interrupt that fires when data is received [is disabled](https://github.com/someweisguy/esp_dmx/blob/release/v3.0/src/esp_dmx.c#L1186). The reason for this is because I found I was having issues when sending DMX with the UART receive interrupts enabled so I just disabled them entirely when data is sent. I now see that the reason I needed to disable the UART receive interrupt at all is due to a hardware error in my DMX transceiver hardware. Essentially, my DMX hardware is always receiving data, even when it is sending data. This causes it to receive it's own output, which causes errors. Oops.

I've published a branch called `bugfix/uart-rx-intr` which might fix this issue. I've simply removed all lines in `esp_dmx.c` which disable the UART receive interrupts so that the receive interrupts are active at all times. Hopefully, this fixes your DMX flicker issues. I will be doing some more work on this before I merge the changes to the main branch.

I think your use case will still result in some packet jitter when receiving and sending data on the same time from one driver but I am not positive about that.

someweisguy, 2023-02-03:
I've finally fixed my hardware issues and tested the new branch. Everything looks good on my end so I submitted and merged a pull request. I am not able to test your exact setup. Since I haven't heard from you in a while I am going to assume that this issue is resolved. Let me know if there are any other issues!

closed 2023-02-03 by someweisguy