someweisguy/esp_dmx #39: esp flash dump error on

surajchauhan6631, 2023-01-23:
when I upload DMXRead example of esp_dmx library on my esp32 module and open serial monitor for getting the status of DMX. I am getting this output on serial monitor

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (131) esp_core_dump_flas���J����ɕ�сsize of core dump image: 151653755

what does it means??.

someweisguy, 2023-01-24:
I could be mistaken here, but these look like the standard ESP32 boot messages. The only thing that I think is unusual is that last line `E (131) esp_core_dump_flas����J����ɕ�сsize of core dump image: 151653755`.

I don't believe these messages have anything to do with esp_dmx. You should be able to make most, if not all, of these messages go away by connecting GPIO15 to ground.

closed 2023-02-03 by someweisguy

niklasberlin, 2023-02-21:
for future Reference and people who find this issue because they have a similar problem:

I had a similar Problem when i had my DMX circuit connected to pins of my ESP32 that are shared with the internal flash and when i tried to manipulate those pins with my code, the ESP was unable to access the flash correctly -> after moving the connections to different pins the ESP booted as expected

for reference on pin assignment i used the following pinout diagram:
https://www.mischianti.org/2021/05/26/esp32-wroom-32-high-resolution-pinout-and-specs/