someweisguy/esp_dmx #40: does not (seem to) recieve data

Theexternaldisk, 2023-01-23:
I can't seem t recieve DMX data. I'm using a DOIT EPS32 DEVKIT V1 and a RS485 to TTL module (https://www.elecrow.com/uart-ttl-to-rs485-twoway-converter-p-1545.html)
The module RX pin is connected to GPIO 16 and the RX LED is flickering when a DMX signal is applied.
The DMX signal is provided by a simple DMX controller (Elation SDC-16).
What I've tried:
 - both the 3.0 and 2.0.2 release of the library with the standard DMXread sketch. Both don't get any data. Only get an error when disconnecting and connecting the DMX signal.
 - both uart port 1 and 2, without succes.
 - connecting two RS485 modules A and B together with one module on the RX pin and the other on the TX pin and using the arduino multiserial example to send and recieve data. This worked fine even with the baudrate set to 2500000.
 - swapping RS485 modules, without succes

haven't tried:
- looking at the DMX signal on a scope
- using other GPIO pins for the RX TX pins
- diving deeper into the library to see what is going on

any ideas?

Theexternaldisk, 2023-01-24:
Hi,
No I'm not the same person as on reddit. 
Sorry forgot to add the schematic I found of this module: https://webshop.domoticx.nl/image/catalog/modbus/RS485%20naar%20RX-TX%20TTL%203.3V%205V%20omvormer%20module%20SP3485/RS485%20naar%20RX-TX%20TTL%203.3V%205V%20omvormer%20module%20SP3485%20schema.png 
The module is set to be in read mode while not transmitting data.
Also forgot to say that I've got two of these working one as a transmitter on the ESP32 DEVKIT and one as a reciever on an ESP-C3 mini. Which works flawlessly, it sends and recieves data without any problem.
I'll try hooking it up to an oscilloscope tomorrow.

Theexternaldisk, 2023-01-25:
ohh man, I'm such an amateur :)
Turns out the A or B line was shorted to the DMX ground....
Soldered my own PCB to hold all the modules I need but must have mist this one solder bridge.
Well great work on the library works like a charm.

someweisguy, 2023-01-26:
I'm glad you got it figured out! Hardware is difficult. I'm currently waiting on v4 of my PCB to come in because I messed up the previous three so far. D'oh.

Let me know if you have any other questions or concerns!

closed 2023-01-26 by someweisguy