someweisguy/esp_dmx #43: ESP32-WROOM-32 together with MAX485 module

blockiyt, 2023-02-09:
Hi, first of all, this seems to be an up to date libary which is very good, I couldn't another one.
The Problem is that I don't know how to wire and program it, because I am very new in that scene, so I would ask politely if someone could help me with that.
Thanks!

someweisguy, 2023-02-10:
Thanks for the question!

In this library's readme, there is a wiring guide that can be found [here](https://github.com/someweisguy/esp_dmx#wiring-an-rs-485-circuit). The simplest possible way that I've found to put a DMX device together is not to wire your own circuit, but to purchase the [Adafruit ESP32 Feather Board](https://www.adafruit.com/product/3405) and the [Sparkfun DMX Shield](https://www.sparkfun.com/products/15110). If you don't have a DMX device that you can control, I would recommend buying two of each module so that you can program one ESP32 to send data, and one ESP32 to receive data. 

As far as programming goes, I would recommend that you use Arduino. There is more information in the readme about using this library on Arduino [here](https://github.com/someweisguy/esp_dmx#arduino). I've written several example sketches in this library for Arduino that are well-commented to explain what each line of code is doing. :)

Let me know if that answers your questions, and if you have any other questions!

blockiyt, 2023-02-13:
Yeah, I think that solved my problem, thank you!
I bought the boards and shields you mentioned above and I'll try to get it working, I just thought I could use the MAX because I had it laying around :)

closed 2023-02-13 by blockiyt