someweisguy/esp_dmx #45: Problem with library installation for PlatformIO

zms16, 2023-02-13:
Hello. I'm trying to install this library for PlatformIO ESP-IDF project. Sorry if my problem is not exactly related to this library, but I don't know what to do now. I did a clean reinstall of VSCode, PlatformIO and started a new ESP-IDF project to ensure there are no legacy code conflicts.

1. Installation by PlatformIO Library Register gives me this error:
> Error: Could not find the package with 'someweisguy/esp_dmx @ ^3.0.0-beta' requirements for your system 'windows_amd64'
Deleting "lib_deps = someweisguy/esp_dmx@^3.0.0-beta" from **platformio.ini** allows me to build the project, but I get several errors in **dmx_hal.h**:
> too few arguments to function 'uart_ll_set_baudrate'
> too few arguments to function 'uart_ll_get_baudrate'
> control reaches end of non-void function [-Werror=return-type]
> too few arguments to function 'uart_ll_set_baudrate'
> 'uart_dev_t' has no member named 'uart_conf0_reg_t'
> 'uart_dev_t' has no member named 'uart_conf0_reg_t'
> control reaches end of non-void function [-Werror=return-type]
> 'uart_dev_t' has no member named 'uart_status_reg_t'
> control reaches end of non-void function [-Werror=return-type]
And one in **esp_dmx.h**:
> implicit declaration of function 'esp_timer_get_time' [-Werror=implicit-function-declaration]

2. Cloning "esp_dmx-release-v3.0" folder into **/components** folder of my project causes **#include "esp_dmx.h"** error.

3. Cloning  "esp_dmx-release-v3.0" folder into **/lib** folder causes last five **dmx_hal.h** errors from 1. but no other errors.

Are there any other ways to install a library to PlatformIO ESP-IDF project?

someweisguy, 2023-02-13:
Thanks for the feedback!

I'm not sure what is causing this, but I have some guesses. Which ESP32 are you using? The base ESP32? Or something like the ESP32-S3? And what version of ESP-IDF are you using?

someweisguy, 2023-02-14:
I believe I have found the primary issue. The v3.0.0-beta tagged version of this library did not support ESP-IDF v5, which PlatformIO now uses!

I have added ESP-IDF v5 support as of January, so it should be a pretty straightforward fix - just need to publish a new tagged version. However, I am now seeing that PlatformIO inexplicably does not include certain files from ESP-IDF v5. So PlatformIO says it uses ESP-IDF v5, but it's not _actually_ v5. So I am going to add some fixes for this issue and I'll publish a new tagged version ASAP!

If you could still verify that you're using ESP-IDF v5 on PlatformIO and which ESP32 hardware you're using, that would be excellent!

zms16, 2023-02-14:
Yes, I'm using ESP32-S3. And the latest ESP-IDF version, I guess it is v5.0 according to changelog, I don't know how to check it in VSCode.
I had some of the features missing after clean VSCode reinstall too. It seems that reinstalling and just updating is not the same.
Thanks for reply and upcoming fix!

someweisguy, 2023-02-15:
This appears to be fixed! I published v3.0.1-beta with the new changes. 

BUT: I was a dummy and published the PlatformIO package before testing for ESP32-S3. So for right now, you will have to clone this repo into your `lib` folder on PIO. I am waiting for the package to become unpublish on PIO. Once it is, I will republish it with the fixes for ESP32-S3, and close this issue. 

Let me know if you have any issues!

Bombenscheck, 2023-02-15:
hello, I am trying to use the Library in an arduino Projekt in Platformio.
Project configuration:

> [env:FirstFlash]
> platform = espressif32
> board = esp32doit-devkit-v1
> framework = arduino
> monitor_speed = 115200
> upload_speed = 115200
> board_build.filesystem = littlefs

Is the Library Arduino compatible?
I installed the new version in the lib folder as described but I get a boatload of compiling errors similar to these ones:

> lib/esp_dmx/src/rdm_types.h:238:3: error: conflicting types for 'rdm_disc_unique_branch_t'
>  } rdm_disc_unique_branch_t;
>    ^~~~~~~~~~~~~~~~~~~~~~~~
> In file included from lib/esp_dmx/src/esp_rdm.h:8,
>                  from lib/esp_dmx/src/esp_dmx.h:15,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.h:7,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.c:1:
> lib/esp_dmx/src/rdm_types.h:238:3: note: previous declaration of 'rdm_disc_unique_branch_t' was here
>  } rdm_disc_unique_branch_t;
>    ^~~~~~~~~~~~~~~~~~~~~~~~
> In file included from lib/esp_dmx/src/private/rdm_encode/functions.h:9,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.c:1:
> lib/esp_dmx/src/rdm_types.h:245:16: error: redefinition of 'struct rdm_disc_mute_t'
>  typedef struct rdm_disc_mute_t {
>                 ^~~~~~~~~~~~~~~
> In file included from lib/esp_dmx/src/esp_rdm.h:8,
>                  from lib/esp_dmx/src/esp_dmx.h:15,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.h:7,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.c:1:
> lib/esp_dmx/src/rdm_types.h:245:16: note: originally defined here
>  typedef struct rdm_disc_mute_t {
>                 ^~~~~~~~~~~~~~~
> In file included from lib/esp_dmx/src/private/rdm_encode/functions.h:9,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.c:1:
> lib/esp_dmx/src/rdm_types.h:251:3: error: conflicting types for 'rdm_disc_mute_t'
>  } rdm_disc_mute_t;
>    ^~~~~~~~~~~~~~~
> In file included from lib/esp_dmx/src/esp_rdm.h:8,
>                  from lib/esp_dmx/src/esp_dmx.h:15,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.h:7,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.c:1:
> lib/esp_dmx/src/rdm_types.h:251:3: note: previous declaration of 'rdm_disc_mute_t' was here
>  } rdm_disc_mute_t;
>    ^~~~~~~~~~~~~~~
> In file included from lib/esp_dmx/src/private/rdm_encode/functions.h:9,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.c:1:
> lib/esp_dmx/src/rdm_types.h:256:16: error: redefinition of 'struct rdm_device_info_t'
>  typedef struct rdm_device_info_t {
>                 ^~~~~~~~~~~~~~~~~
> In file included from lib/esp_dmx/src/esp_rdm.h:8,
>                  from lib/esp_dmx/src/esp_dmx.h:15,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.h:7,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.c:1:
> lib/esp_dmx/src/rdm_types.h:256:16: note: originally defined here
>  typedef struct rdm_device_info_t {
>                 ^~~~~~~~~~~~~~~~~
> *** [.pio/build/FirstFlash/lib7f7/esp_dmx/esp_dmx.c.o] Error 1
> *** [.pio/build/FirstFlash/lib7f7/esp_dmx/esp_rdm.c.o] Error 1
> In file included from lib/esp_dmx/src/private/rdm_encode/functions.h:9,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.c:1:
> lib/esp_dmx/src/rdm_types.h:269:3: error: conflicting types for 'rdm_device_info_t'
>  } rdm_device_info_t;
>    ^~~~~~~~~~~~~~~~~
> In file included from lib/esp_dmx/src/esp_rdm.h:8,
>                  from lib/esp_dmx/src/esp_dmx.h:15,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.h:7,
>                  from lib/esp_dmx/src/private/rdm_encode/functions.c:1:
> lib/esp_dmx/src/rdm_types.h:269:3: note: previous declaration of 'rdm_device_info_t' was here
>  } rdm_device_info_t

in main.cpp i only imported the header esp_dmx.h do I need to do something with it in the code?

I am not shure if my Problems belong in this issiue or if I should open a new one, please Advise.

similar error messages apear while using Version 3.0.0-beta downloaded over PlatformIO

zms16, 2023-02-15:
Thank you! The code builds perfectly without any errors now. Unfortunately, I can't test it on a esp32-s3 board right now, because there is some problem with uploading after VSCode reinstall, but it is not library related, so I have no more questions, thanks again.

someweisguy, 2023-02-15:
Hi Bombenscheck,

It appears to work on Arduino for me. You should be able to use v3.0.1-beta by cloning it from git into your `lib` folder. Let me know if that works for you. 

It appears PlatformIO has unpublished the mistake I made but I am still having issues publishing fixes. I am hoping this issue will be resolved soon.

Bombenscheck, 2023-02-15:
OK, I Removed all Files from my lib Folder and imported it again. Now it works... I have no Idea what went wrong the first time around. 
Thanks @someweisguy  for your answer.

someweisguy, 2023-02-17:
This should be resolved, so I am going to close this issue.

It turns out that when you ask PlatformIO to unpublish your release, they don't let you republish the release later with the same version number. So I had to publish v3.0.2-beta. You should now be able to add this library as a dependency using the PIO dependency manager.

Let me know if you run into any more issues and I will reopen this ticket!

closed 2023-02-17 by someweisguy