someweisguy/esp_dmx #51: Does not recieve data with less than 512 channels

zms16, 2023-03-06:
Hello again. I've managed to upload the library but stumbed upon a new complication. It seems that my esp32 can't see dmx data with less than 512 channels. I tried 54Ch and 16Ch DMX Controllers, esp32 read nothing on both. But my DIY 512Ch controller works fine and esp32 properly reads 513 bytes. I checked the 54Ch serial on a oscilloscope, it has 44us frame, about 60us break, 20us MAB and almost 8ms between packets.

The code:
```
#include "esp_dmx.h"
#include "esp_log.h"
#include "esp_system.h"
#include "freertos/task.h"

#define RX_PIN 18
#define TX_PIN 17
#define EN_PIN 38

uint8_t dmx_data[DMX_PACKET_SIZE] = {};

void app_main() { 
	const dmx_port_t dmx_num = DMX_NUM_2;
	ESP_ERROR_CHECK(dmx_set_pin(dmx_num, TX_PIN, RX_PIN, EN_PIN));
	ESP_ERROR_CHECK(dmx_driver_install(dmx_num, DMX_DEFAULT_INTR_FLAGS));
	dmx_packet_t packet;
	TickType_t last_update = xTaskGetTickCount();
	while(true) {
		if(dmx_receive(dmx_num, &packet, DMX_TIMEOUT_TICK)) {
			const TickType_t now = xTaskGetTickCount();	
			if (now - last_update >= pdMS_TO_TICKS(1000)) {
				dmx_read(dmx_num, dmx_data, DMX_PACKET_SIZE);
				ESP_LOGI("DMX", "Start code: %02x, Size: %i", packet.sc, packet.size);
				ESP_LOG_BUFFER_HEX("DMX", dmx_data, 16);  // Log first 16 bytes
				last_update = now;
			}
		}
	}
}
```
I'm using:
ESP32-S3 
PlatformIO 3.0.0
ESP-IDF 5.0.0
esp_dmx 3.0.2-beta

userbogd, 2023-03-07:
First of all, i would like to express gratitude for your great work on so useful code! I hope that project will grow and developing! 
I have the same or similar issue. Receiver can receive only fixed number of channels, defined by DMX_MAX_PACKET_SIZE or small less than this value.  
Checked on FTDI DMX master from this project http://hoelscher-hi.de/hendrik/english/openrdm.htm and on this device https://www.recordcase.de/en/eurolite-dmx-led-operator-4. 
I found the commit, which changed the receiver behaviour. This is 54eebcd 'clean uart rx isr codesmell'.I tried to find the critical changing between 'working' ed34948 and 'no-working' 54eebcd commits, but there are a lot of code changed in one commit, and my skills not allow me to find exact reason of this changing. 
I'm not sure that new behaviour is abnormal, but the problem that is no possible to know in advance the number of channels in the packet from various masters. For my opinion the decision of the packet received must be done not on the packet expected size, but on timeout of the next start bit in the flow.
Another one time great thanks for this library! I hope that the suitable solution will be found! I'm ready to continue testing it and provide any help that i can!

someweisguy, 2023-03-07:
Thanks for the great bug report. And thanks @userbogd for the great find!

This is an interesting feature that I tried to ensure was working with this library. When using this library, the ESP32 driver waits to receive 513 bytes of data from the DMX transmitter. When it receives 513 bytes, it sends a notification to the task awaiting for a full packet to be received, if there is a task waiting. In other words, the DMX driver will wait until it receives 513 bytes, and it will notify the user when it does.

The issue that you are both aware of is that DMX packets don't have a minimum packet size. Technically, a DMX packet could contain just a break, mark-after-break, start code and no data. So I've (attempted to) account for this possibility. By default, the DMX driver expects 513 bytes. If it receives fewer than that amount of bytes and then another DMX break, it will notify the user that a full packet has been received and then update its expected packet size to the new, smaller size until then. In other words, the first packet notification comes a little bit later than normal. But then every packet notification afterwards arrives on time. 

That is the desired behavior for normal DMX packets. For RDM packets it works a bit differently because RDM packets contain a byte which includes the packet's size.

Commit https://github.com/someweisguy/esp_dmx/commit/54eebcd91ba3e0078b2f5a9dfc9ff3d038a5dbf9 was supposed to be a refactor of the DMX receive logic which handles those notifications. So it makes sense that this commit is causing the issues. The code before this commit was rather repetitive which many sections of code copy/pasted from other sections so it needed to be cleaned a bit. I'll fix this as soon as I can!

userbogd, 2023-03-07:
> If it receives fewer than that amount of bytes and then another DMX break, it will notify the user that a full packet has been received and then update its expected packet size to the new, smaller size until 

This makes clear for me the mechanism of variable length packet capturing! And this algorithm looks very suitable for this task.

someweisguy, 2023-03-08:
I haven't had a chance to look at this yet, but the relevant lives of code are [here](https://github.com/someweisguy/esp_dmx/blob/release/v3.0/src/esp_dmx.c#L135-L139).

My best guess is that it has something to do with the `driver->end_of_packet` variable which is supposed to evaluate to true if the driver thinks it has reached the end of the DMX packet. More to come soon!

userbogd, 2023-03-08:
The block of code that updates driver->data.rx_size for smaller next expected packet executed normally, so the problem not in driver->end_of_packet state. But the new value of packet size seems not correct, because does not take into account the first start slot. Of cause, it is only my suppose. But the next changing in the code makes it working again;) 
`driver->data.rx_size = driver->data.head - 1;`

someweisguy, 2023-03-09:
An off-by-one error? I'm so embarrassed! :)

Thank you both! I've merged PR #53. Let me know if you find any other issues.

closed 2023-03-09 by someweisguy