someweisguy/esp_dmx #52: Can we transmit 2 DMX Universe ?

brainstormtec, 2023-03-06:
Hello, great work. I have a question.  Can we transmit 2 DMX universe using different UART of ESP32 ?

someweisguy, 2023-03-06:
Thanks for the kind words! Yes, this library can use multiple UART ports to transmit more than one DMX universe. The first argument of nearly all functions in this library is the `dmx_num` which specifies which UART port to use.

Let me know if you have any other questions!

brainstormtec, 2023-03-07:
Thank you so much for your quick response. please further clearify that by using this library will be able to transmit two different DMX universe simultaneously on two different UARTs of ESP32. how to use "dmx_num" to achieve it.

someweisguy, 2023-03-07:
The `dmx_num` I'm referring to identifies which UART port on the ESP32 to use. The ESP32 has several UART ports. The base ESP32 has 3. By default, UART0 is allocated for the serial monitor on your computer. This is how the ESP32 logs messages back to your computer. UART1 and UART2 are unallocated so you are able to use them as DMX ports.

When you install your DMX driver on UART1, you call the function `dmx_driver_install(DMX_NUM_1, DMX_DEFAULT_INTR_FLAGS)`. To install an additional DMX driver on UART2, you can call `dmx_driver_install(DMX_NUM_2, DMX_DEFAULT_INTR_FLAGS)`.

This pattern follows with the remaining functions in the API. To send a packet of DMX data on UART1, you would call `dmx_send(DMX_NUM_1, DMX_PACKET_SIZE)`. To send on UART2, you would call `dmx_send(DMX_NUM_2, DMX_PACKET_SIZE)`.

Hope this helps!

brainstormtec, 2023-03-08:
Thank-you so much for your kind guidance. i will try it.

closed 2023-03-08 by brainstormtec