someweisguy/esp_dmx #55: FORCE_INLINE_ATTR does not name a type

netmindz, 2023-03-23:
Getting this error when trying to use use with WLED as I'm trying to revamp the DMX support to use your more advanced library rather than sparkfun/SparkFunDMX

netmindz, 2023-03-23:
WLED is still on 1.0.6 for ardunio-esp32 and can't easily upgrade.

Would you advise against trying to use your older version and hold off until when/if WLED update?

closed 2023-03-23 by netmindz

someweisguy, 2023-03-23:
I see you just closed this, but I thought I'd answer anyway!

It's up to you if you want to use the older versions of this library. I am not supporting any of the older versions of this library anymore, so you'd just have to deal with any bugs that you may encounter. 

Arduino-ESP32 v2.0.0 has been out for nearly two years now, so, without knowing about the WLED library you're using, I'm not sure if WLED will ever support v2. The reason why I require v2 is because it supports a few of the ESP32 variants, like the ESP32-S2 and ESP32-S3.