someweisguy/esp_dmx #57: dmx recieve buffer shifts after saving data to FAT

zms16, 2023-03-23:
Hello. I found another issue. I remember reading an issue about data shifting in dmx recieve buffer but I couldn't find a closed issue so I decided to start a new one.
I'm making a dmx recorder, using two RS-485 chips, one for recieve and one for transmit. It reads dmx input from the rx line and instantly sends it unchanged to the tx line. If a user wants to record, he or she presses a button and the device copies dmx recieve buffer each 60ms to a large buffer which allocated in a PSRAM. When a user finished recording, the device saves the large buffer to spi flash through FAT filesystem with ESP wear leveling library. The process is long, it takes several seconds to save data to FAT and it blocks the code.

The problem is, after that save the dmx recieve buffer is shifted. The first channel could be on 4th or 35th channel now. The program works as intended before the save and when the device is playing recorded data (opening FAT file doesn't shift data). The size of saved data doesn't matter, the buffer shifts anyway. 
I'm also using NVS to save some meta information variables along with FAT save, so it can be the issue too.

To clarify, this is my dmx code:
```
uint8_t dmx_data[DMX_PACKET_SIZE] = {};
uint8_t dmx_data_out[DMX_PACKET_SIZE] = {};
if(dmx_receive(RX_DMX, &packet, 0)) {
	dmx_read(RX_DMX, dmx_data, DMX_PACKET_SIZE);
}
dmx_write(TX_DMX, dmx_data_out, DMX_PACKET_SIZE);
dmx_send(TX_DMX, DMX_PACKET_SIZE);
dmx_wait_sent(TX_DMX, DMX_TIMEOUT_TICK);
```
dmx_data is my recieve buffer and dmx_data_out is my transmit buffer.

I'll do some tests tomorrow to find whether FAT or NVS is the issue, provide more information and a full shortened program code if it helps.

zms16, 2023-03-24:
```
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "driver/uart.h"
#include "driver/gpio.h"
#include "nvs_flash.h"
#include "nvs.h"
#include "esp_log.h"
#include "esp_vfs.h"
#include "esp_vfs_fat.h"
#include "esp_system.h"
#include "esp_dmx.h"

#define RX_PIN 18
#define TX_PIN 17
#define RX_DMX DMX_NUM_2
#define TX_DMX DMX_NUM_1

uint8_t dmx_data[DMX_PACKET_SIZE] = {};
uint8_t j = 0;

static wl_handle_t s_wl_handle = WL_INVALID_HANDLE; // Handle of the wear levelling library instance
const char *base_path = "/spiflash"; // Mount path for the partition

void app_main() { 
	uint8_t* dmx_buff = (uint8_t*) malloc(100*513*sizeof(uint8_t));	
//------DMX INIT---------------------------------------------
	ESP_ERROR_CHECK(dmx_set_pin(RX_DMX, DMX_PIN_NO_CHANGE, RX_PIN, DMX_PIN_NO_CHANGE));
	ESP_ERROR_CHECK(dmx_driver_install(RX_DMX, DMX_DEFAULT_INTR_FLAGS));
	ESP_ERROR_CHECK(dmx_set_pin(TX_DMX, TX_PIN, DMX_PIN_NO_CHANGE, DMX_PIN_NO_CHANGE));
	ESP_ERROR_CHECK(dmx_driver_install(TX_DMX, DMX_DEFAULT_INTR_FLAGS));
	dmx_packet_t packet;
//------NVS INIT---------------------------------------------
	nvs_handle_t nvs_handle;
    esp_err_t err = nvs_flash_init_partition("nvs_storage");
	ESP_LOGI("NVS", "init: %s", esp_err_to_name(err));
	if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
		nvs_flash_erase_partition("nvs_storage");
		err = nvs_flash_init_partition("nvs_storage");
		ESP_LOGI("NVS", "re-init: %s", esp_err_to_name(err));
	}	
	err = nvs_open_from_partition("nvs_storage","DMX", NVS_READWRITE, &nvs_handle);
	ESP_LOGI("NVS", "open: %s", esp_err_to_name(err));
//------FAT INIT---------------------------------------------
	size_t f_bytes = 0; 
	const esp_vfs_fat_mount_config_t mount_config = {
		.max_files = 6,
		.format_if_mount_failed = true,
		.allocation_unit_size = CONFIG_WL_SECTOR_SIZE
    };
	ESP_LOGI("FAT", "Mounting FAT filesystem");
	err = esp_vfs_fat_spiflash_mount_rw_wl(base_path, "fat_storage", &mount_config, &s_wl_handle);
	if (err != ESP_OK) {
        ESP_LOGE("FAT", "Failed to mount FATFS (%s)", esp_err_to_name(err));
    }

    while(true) {	
		if(dmx_receive(RX_DMX, &packet, 0)) {
			dmx_read(RX_DMX, dmx_data, DMX_PACKET_SIZE);
		}
		dmx_write(TX_DMX, dmx_data, DMX_PACKET_SIZE);
		dmx_send(TX_DMX, DMX_PACKET_SIZE);
		dmx_wait_sent(TX_DMX, DMX_TIMEOUT_TICK);

		memcpy((dmx_buff+j*513),dmx_data,513);
		++j;
		if(j==100) {
			FILE *f = fopen("/spiflash/file1.bin", "wb");
			if (f == NULL) {
				ESP_LOGE("FAT", "Failed to open file for writing");
			}
			f_bytes = fwrite(dmx_buff, sizeof(uint8_t), 513*100, f);
			fclose(f);
			ESP_LOGI("FAT", "%zu bytes written", f_bytes);
			j=0;
		}
		vTaskDelay(30 / portTICK_PERIOD_MS);
	}
}

```

zms16, 2023-03-24:
So I made a program that still causes recieve buffer to shift. It is not NVS, the shift is caused by FAT save.

someweisguy, 2023-03-24:
Thanks for the example code! Which version of ESP-IDF are you using?

zms16, 2023-03-27:
Yes, I'm using PlatformIO, ESP-IDF 5.0.0 and ESP32-S3

someweisguy, 2023-03-27:
Is the esp_dmx library folder in your `.vscode` folder? If so, could you make a folder called `components` in your project directory, and move the esp_dmx folder to the `components` folder? The end result should be something like `my_dmx_recorder_project/components/esp_dmx/` if your project is called `my_dmx_recorder_project`. 

What this should do is allow the ESP32 build system to find the `Kconfig` file in esp_dmx. This file interfaces with the ESP32's menuconfig. It will instruct the ESP32 to put the DMX driver and hardware timer driver into IRAM, which means that the DMX interrupts will continue to fire even when you are writing to flash. 

That is my best guess of the issue at the moment. I think that you may be writing to flash in the middle of receiving a DMX packet. Because reading from or writing to flash stops the DMX interrupts, it is causing the DMX driver to miscount which byte corresponds to which DMX address which leads to the buffer shift.

If this fixes the issue, I will also add a function which can momentarily disable the DMX receive interrupts so that you can safely read or write to flash even if the DMX driver is not in IRAM.

zms16, 2023-03-28:
The library was in the `/lib` folder, but moving esp_dmx to `/components` still helped, thank you.

closed 2023-03-28 by zms16

someweisguy, 2023-04-01:
I'm working on #60 to fix this problem as well. With the added changes in this PR, you are able to call `dmx_driver_disable()` before reading or writing to flash. This disables the DMX driver gracefully, meaning that the driver will stop processing incoming DMX data and calls to `dmx_send()` and `dmx_receive()` are disabled. Any function that relies on these functions, such as any function which sends or receives RDM, will fail as well. When you are done with your work in flash, calling `dmx_driver_enable()` will reenable the DMX driver, allowing it to send and receive DMX data once again.

Disabling the driver is not needed when placing the DMX driver in IRAM. Placing the DMX driver in IRAM is the ideal solution, but I understand there are some situations where doing so isn't possible.