someweisguy/esp_dmx #62: [question] Not Receiving data, wrong pinout, or incompatible module?

cnf, 2023-05-02:
Hi,

I seem to be having a similar issue as https://github.com/someweisguy/esp_dmx/issues/40.

I have an [M5Stick-C](https://docs.m5stack.com/en/core/m5stickc) with a [M5 RS485 Hat](https://docs.m5stack.com/en/hat/hat-rs485). The [M5 Arduino example](https://github.com/m5stack/M5StickC/blob/master/examples/Hat/RS485/RS485.ino) show me data, and I can see the DMX channel values, but with your library, it doesn’t seem to receive anything.

I suspect it is because this module doesn’t have a `RTS` pin, and manages that automatically, so I was hoping your equivalent modules arrived, and you could help me out with this.

I am trying to receive only, and the data is being sent from an ENTTEC usb DMX sender, and QLCPlus.

[image]

Thanks!

someweisguy, 2023-05-04:
Thanks for reaching out!

As you probably read, the issue in #40 was a hardware error where something was shorted to ground. But since you are able to read data using `Serial2.read()` I'm guessing it isn't a hardware error. Do I understand correctly that you are initializing your `Serial2` to 250kbaud, 2 stop bits, no parity? This library automatically initializes the serial port to that configuration so it could explain why you aren't receiving data using esp_dmx.

If that is not the issue, can you confirm for me which version of esp_dmx you are using? And are you using Arduino?

cnf, 2023-05-05:
Hi,

I have tried with and without `Serial2.begin(250000, SERIAL_8N2, 26, 0);`

I am running https://github.com/someweisguy/esp_dmx/blob/release/v3.0/examples/DMXRead/DMXRead.ino almost verbatim, but in a platformIO project, and with M5 specifics sprinkled in (mostly to use the LCD screen on it.)

`dmx_receive(dmxPort, &packet, DMX_TIMEOUT_TICK);` always returns 0, no matter what I do.

I am using `someweisguy/esp_dmx@^3.0.2-beta`

https://gist.github.com/cnf/36ff0779ba7639840cc90047da6f78ca is my latest try.

Thanks!

someweisguy, 2023-05-10:
Thanks for your patience on this. I admit, I'm a bit stumped. I just published v3.0.3-beta which contains fixes to a few issues. It should be live on the Arduino Library manager in a few hours (no more than 24 hours). Perhaps you can try to update to the new version to see if one of the fixes has solved this problem?

cnf, 2023-05-10:
Ha, I am stumped as well, i’ll give that a go and report back… I am also open to the idea that the M5 RS485 module has issues, though that would be sad :/

someweisguy, 2023-05-10:
Upon rereading, #51, I'm feeling somewhat confident that this is the issue you are experiencing. In short, if you received fewer than 512 addresses of DMX data, the driver would not receive any data whatsoever. If this is indeed the issue you are experiencing, it should be fixed in v3.0.3-beta!

cnf, 2023-05-10:
with v3.0.3-beta, it seems to be the same…
I am making sure to even send data on channel 1, and channel 512
Anything I can do to debug this?

cnf, 2023-05-10:
I just found out M5 has an actual DMX addon for their M5 core modules, with the same chipset:
https://docs.m5stack.com/en/module/DMX_Base

it seems to be wired the same, AND they use this library to drive it... so it should work? Reading their code to see if I am doing something wrong...

someweisguy, 2023-05-10:
Good find! It looks like M5 uses v2 of this library. Then something that _should_ work would be to revert back to esp_dmx v2 and try the [DMXRead](https://github.com/someweisguy/esp_dmx/blob/release/v2.0/examples/DMXRead/DMXRead.ino) example. 

If you are still not able to read data using v2 of the library I would think that there is a hardware issue somewhere.

cnf, 2023-05-10:
Ok, I am getting output!

```
Reconnecting to COM5     Connected!
+Start code is 0x00 and slot 1 is 169
+++++++++++Start code is 0x00 and slot 1 is 169
+++++++++++Start code is 0x00 and slot 1 is 169
++++++++++++Start code is 0x00 and slot 1 is 169
+++++++++++Start code is 0x00 and slot 1 is 169
+++++++++++++Start code is 0x00 and slot 1 is 169
+++++++++++Start code is 0x00 and slot 1 is 169
++++++++++++Start code is 0x00 and slot 1 is 169
++++++++++++Start code is 0x00 and slot 1 is 169
```

what did it was `dmx_set_pin(dmxPort, -1, receivePin, -1);`
set `TX` and `EN` to -1

So now I am getting data, now I can try and figure out the rest :P

Thanks for your patience so far :P

someweisguy, 2023-05-10:
Fascinating! I wouldn't think that would solve the problem. Good work with this solution. :)

I'll close this for now. Let me know if you have any more questions or issues!

closed 2023-05-10 by someweisguy