someweisguy/esp_dmx #65: Single Core Issues?

jono-chandler, 2023-06-05:
Hey!

For context:
The board I am using: ES32-C3-WROOM-02 (Single core rather than dual-core)
IDE: Arduino

After loading the DMXWrite Example, and adapting the pins to TX=20, RX=21, Enable=0 (I believe this those are the ports for UART port 1 on this board?) -- and uploading, the ESP32 is consistently reboots, and gives this serial message:

`ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381a00
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x91c
load:0x403ce710,len:0x25b0
entry 0x403cc710`

Other code/different library examples upload and run fine

Does the library (in the default configuration) require timers/interrupts that aren't within this board/only in 2 core varieties? Or am I missing something obvious?

Any ideas/info would be great!
Thanks!

someweisguy, 2023-06-07:
Thanks for bringing this to my attention! I've loaded up my ESP32C3 on Arduino and I am seeing this same error. I am not seeing this error on ESP-IDF, however, so that indicates to me that something funky is going on with Arduino for the ESP32-C3.

I will investigate and keep you updated!

someweisguy, 2023-06-23:
Thanks for your patience on this. I have found the issue and it is two-fold. 

First there was an issue with the DMX driver attempting to use one of the ESP32C3's hardware timers that was reserved for the Watchdog. Because the watchdog timer was getting clobbered by the DMX driver, this was causing the ESP32C3 to hang upon calling `dmx_driver_install()`.

Second, there appears to be an issue with the pins that you are assigning for use with the DMX driver. Pins 20 and 21 are indeed labelled TX and RX on the ESP32C3 dev board, but these are for UART 0, which is used for communication to and from the Serial Monitor. When giving these pins to the DMX driver by calling `dmx_set_pin()`, you are clobbering the Serial Monitor pins. I believe that this should be fine if you don't need to read or write to the Serial Monitor. If you are using the Serial Monitor for logging messages, you will need to use different pins. 

I've submitted PR #69 to fix this issue. I will be merging it shortly. When I do, you are able to clone the head of the `release/v3.0` and you will be able to use this library. Let me know if you have any questions!

closed 2023-06-23 by someweisguy