someweisguy/esp_dmx #75: compile example DMXwrite.ino failed

everyb313, 2023-06-29:
compiling the DMXwrite example failed. 
Win10 / arduino IDE 2.10

....
C:\\Users\\Hagen\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\esp-2021r2-patch5-8.4.0/bin/xtensa-
C:\Users\Hagen\Documents\Arduino\esp_dmx-release-v3.0\examples\DMXWrite\DMXWrite.ino: In function 'void setup()':
C:\Users\Hagen\Documents\Arduino\esp_dmx-release-v3.0\examples\DMXWrite\DMXWrite.ino:57:3: error: 'dmx_config_t' was not declared in this scope
   dmx_config_t config = DMX_CONFIG_DEFAULT;
   ^~~~~~~~~~~~
C:\Users\Hagen\Documents\Arduino\esp_dmx-release-v3.0\examples\DMXWrite\DMXWrite.ino:57:3: note: suggested alternative: 'gpio_config_t'
   dmx_config_t config = DMX_CONFIG_DEFAULT;
   ^~~~~~~~~~~~
   gpio_config_t
C:\Users\Hagen\Documents\Arduino\esp_dmx-release-v3.0\examples\DMXWrite\DMXWrite.ino:58:32: error: 'config' was not declared in this scope
   dmx_driver_install(dmxPort, &config, DMX_INTR_FLAGS_DEFAULT);
                                ^~~~~~
C:\Users\Hagen\Documents\Arduino\esp_dmx-release-v3.0\examples\DMXWrite\DMXWrite.ino:58:32: note: suggested alternative: 'confstr'
   dmx_driver_install(dmxPort, &config, DMX_INTR_FLAGS_DEFAULT);
                                ^~~~~~
                                confstr
C:\Users\Hagen\Documents\Arduino\esp_dmx-release-v3.0\examples\DMXWrite\DMXWrite.ino:58:40: error: 'DMX_INTR_FLAGS_DEFAULT' was not declared in this scope
   dmx_driver_install(dmxPort, &config, DMX_INTR_FLAGS_DEFAULT);
                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\Hagen\Documents\Arduino\esp_dmx-release-v3.0\examples\DMXWrite\DMXWrite.ino:58:40: note: suggested alternative: 'ESP_INTR_FLAG_LEVEL4'
   dmx_driver_install(dmxPort, &config, DMX_INTR_FLAGS_DEFAULT);
                                        ^~~~~~~~~~~~~~~~~~~~~~
                                        ESP_INTR_FLAG_LEVEL4

Using library esp_dmx at version 3.0.3-beta in folder: C:\Users\Hagen\Documents\Arduino\libraries\esp_dmx 
exit status 1

Compilation error: 'dmx_config_t' was not declared in this scope

someweisguy, 2023-06-29:
Thanks for reaching out! It appears that you may be using an old version of the library for the base code, but a newer version of the library for the example code. To fix this, you have two options: you can use the release version of the code (the old code) or you can use the unreleased version of the code (the newer, experimental code).

To use the old code, you can install the library using the Arduino library manager and load the DMXWrite example from the Arduino IDE. To use the new code, you can clone this library into your Arduino `libraries` folder and use the updated examples.

Hope this helps! Let me know if you have any more questions!

everyb313, 2023-06-29:
Thanks ! You were right !
I updated to the 3.03 beta but for som reason an old library was used. After fixing this it worked lika charm.
Thanks for your hints !
:-)

closed 2023-06-29 by everyb313