someweisguy/esp_dmx #78: agent.h - error: C99 designator 'software_version_label' outside aggregate initializer

jonasprobst, 2023-07-16:
Get the following Error when I try compiling the library:
```
compiling .pio/build/debug_espressif32/src/main.cpp.o
In file included from .pio/libdeps/debug_espressif32/esp_dmx/src/esp_dmx.h:13,
                 from lib/DMXController/DMXController.h:4,
                 from src/main.cpp:4:
.pio/libdeps/debug_espressif32/esp_dmx/src/dmx/agent.h:34:3: error: C99 designator 'software_version_label' outside aggregate initializer
   }
   ^
lib/DMXController/DMXController.h:109:26: note: in expansion of macro 'DMX_CONFIG_DEFAULT'
   dmx_config_t config_ = DMX_CONFIG_DEFAULT; // The default configuration.
```

The DMX_CONFIG_DEFAULT macro is using a designated initializer to create a dmx_config_t struct with default values. The software_version_label field is being initialized with a string that's constructed using the ESP_DMX_VERSION_MAJOR, ESP_DMX_VERSION_MINOR, and ESP_DMX_VERSION_PATCH macros.

Designated initializers are a feature of the C language that allow you to initialize specific fields of a struct by name. They were introduced in the C99 standard, but not all compilers support them, especially in C++ code.

I'm using PlatformIO in vscode on MacOSX Ventura 13.4.1. The per default installed compiler on PlatformIO for the Espressif 32 Platform is as follows:
```
Name				Type			Requirements		Installed		
toolchain-xtensa-esp32		toolchain		8.4.0+2021r2-patch5	8.4.0+2021r2-patch5
```
So, this should therefore work as GCC 8.4.0 supports the C99 standard. But somehow it doesn't.

Workaround: Create my own "default" settings without the designated initialisers, like so:

```
    static constexpr dmx_config_t config_ = {
        255,  // alloc_size
        0,    // model_id
        0x0100,  // product_category
        ESP_IDF_VERSION_VAL(0, 0, 1),  // static software_version_id
        "esp_dmx v0.0.1",  // static software_version_label
        1,  // current_personality
        {{1, "Default Personality"}},  // personalities
        1,  // personality_count
        0   // dmx_start_address
    };
```
Beeeeeeeyuuuuuuuutiful :-D
Maybe it's just me. Or is it something worth looking into?

someweisguy, 2023-07-17:
Thanks for the detailed feedback! This was a nice find! Issues like this are great at keeping me humble. :P

I'll update the macro using the non-C99 initializers as shown in your example code!

someweisguy, 2023-07-21:
This is fixed in the latest update to the release/v3.0 branch. Let me know if you find any other issues!

closed 2023-07-21 by someweisguy