someweisguy/esp_dmx #79: Can't get the Read Example to work on ESP32 Arduino in PlatformIO

jonasprobst, 2023-07-19:
I really wanna use this library because it's well documented, actively developed and non-blocking. So big **THANK YOU!!**

Unfortunately i can't get the arduino read example to work. Im on a ESP32 Thing Plus with the Sparkfun DMX Shield. 

My platformio.ini
```
[env]
platform = espressif32
monitor_speed = 115200
framework = arduino
lib_deps =
    esp_log.h
    someweisguy/esp_dmx @ 3.0.3-beta
    ;https://github.com/someweisguy/esp_dmx/

[env:debug_thingPlus]
board = sparkfun_esp32s2_thing_plus_c
board_upload.flash_size = 16MB
build_flags = 
    -D CORE_DEBUG_LEVEL=5
```

And then I'm using your example from here: 
https://github.com/someweisguy/esp_dmx/blob/v3.0.3-beta/examples/DMXRead/DMXRead.ino

I've only added an else statement at the end with a debug message, like so:
```
...    
/* Stop the program. */
    while (true) yield();
    
 // This is what i've added:
  } else {
    Serial.println("Waiting for DMX...");
    Serial.println("Error: " + String(packet.err));
  }
```

The packet return the **Error: 263**. In my research a came across this breadcrumb-trail:

> The ESP32 error 263 corresponds to a timeout error, often associated with I2C communication. This error might occur due to several reasons such as:
>
> A bad connection in your I2C setup.
> Incorrect configuration of the I2C setup.
> Not sending the correct register addresses.

Any ideas what's happening?

jonasprobst, 2023-07-19:
By the way, I'm using UART Port1 and GPIO Pins 16, 17 and 21 (standard ESP32) according to their DMX library [example code](https://t.ly/uZG02):

```
... // lines 28 - 30
int enablePin = 21;		//dafault on ESP32
int rxPin = 16;
int txPin = 17;
...
```

Just realised, after studying those products for hours, that there is a discrepancy in the documentation: In their DMX library (above) they'r using GPIO21 as Enable. But if I map the the [Sparkfun DMX Shield documentation](https://t.ly/2th73) to the [Sparkfun ESP32 Thing Plus graphical datasheet](https://t.ly/J9lCE) it looks like Enable is GPIO4. I will ring the board through with my multimeter later today to rule out that it's a hardware / pin definition issue.

someweisguy, 2023-07-19:
Thanks for the kind words and so sorry for the confusion! I am also a bit perplexed - I have a couple Sparkfun DMX shields as well and I've had no issues using GPIO 21 as the enable pin. I do not see a revision number listed on my Sparkfun DMX shields. If your has a revision number on it, then I suppose that would be the issue.

Two quick side notes:

1. If you are using RDM with the Sparkfun shield you should solder a 120ohm resistor as the terminating resistor if you haven't already. Using RDM without that terminating resistor may cause problems sending and receiving RDM requests.
2. Based on the issue you submitted in #78, it appears you are using the latest commit which is currently experimental. You may continue to have issues using the experimental commit, but hopefully they are few and far between. If you would like to use a more stable commit, you should use the tagged commits. That said, I am very nearly done with version 3.1 so hopefully you'll have a stable version with the latest features soon!

Let me know if you have any comments, questions, or concerns!