someweisguy/esp_dmx #82: ESP_DMX on ESP32-S2

someweisguy, 2023-07-23:
### Discussed in https://github.com/someweisguy/esp_dmx/discussions/81

<div type='discussions-op-text'>

<sup>Originally posted by **fedechi** July 21, 2023</sup>
Hi Mitch,
first of all thanks for your work!
I'm trying, in very optimistic vay, to use the library with ESP32-S2 without success (v3.1).
Will the library be compatible with these devices in the future? 
or am I making some mistake.
thanks
Federico</div>

>yeah great work. i'm working on custom board with ESP32-S2FNR4R2 for build responder, for test i use Enttec DMX USB PRO MK2.
I'm sure the hardware works from previous tests.
I use ESP-IDF 4.4.2 and ESP-PROG with jtag for debug and programm. The library version is 3.1.
simply try to execute the example idf_rdm_responder.

>when programming the output console show
Info : esp32s2: Target halted, PC=0x400236D5, debug_reason=00000001

>when i try to debug the PC fall immediatly in panic
panic_abort(const char *details) with detail 97

>if i go back on commit of lib (for example 15 jun 2023)
the software run, i can debug and see DMX packet correctly but RDM not work (no discovery).
sorry, I don't currently have the ability to debug this type of device in depth.
thanks for support
F.

someweisguy, 2023-07-26:
The latest commit on release/v3.1 compiles and awaits RDM data without issue. I haven't tried testing any RDM receiving yet.

fedechi, 2023-07-29:
ok, it was my mistake really sorry. I inherited from the previous examples the initialization in the form
ESP_ERROR_CHECK(dmx_driver_install(dmx_num, &config, DMX_INTR_FLAGS_DEFAULT));
but having changed the return of the function obviously nothing worked anymore!
Now dmx works, the code run but unfortunately I still have problems on RDM discovery with my test setup.
I can't discover with ENTTEC RDM controller, the therminal says
W (20592) dmx: PID 0x0001 did not send a response
With the oscilloscope I can observe that after idle state of DMX line, when responder receive the first RDM_PID_DISC_UNIQUE_BRANCH
to which he must answer, the responder switch to TX mode the dmx line quite early (<176us) but writes nothing. After 25ms the line return to RX state.
Also through RDM packet sniffer i can see an error on checksum for ACK response to DISC_MUTE command.
thanks
F.

someweisguy, 2023-08-07:
Interesting. Thank you for the detailed information!

I will need to do some more in-depth research into this problem. Unfortunately, I am in the middle of moving across the country so I cannot access my testing equipment at the moment.

This library uses a timer to keep track of when the latest DMX/RDM data was received. When the ESP32 receives an RDM request, it will attempt to send a response. It sounds like you are already aware that RDM requires that responses to packets be sent after about 176 microseconds and within about 3 milliseconds. If more than 3ms have elapsed when the library is attempting to send an RDM response packet, this library prints a warning message. This is why you are seeing `W (20592) dmx: PID 0x0001 did not send a response`. It seems like there may be an error in the way this library configures the timer on ESP32-S2.

How many RDM devices do you have connected on your DMX network? It appears you may have at least 3 different devices: the ENTTEC RDM controller, the ESP32-S2, and a different device manufactured by a company called _ERAL srl_. The device manufactured by _ERAL srl_ is the device that is returning a bad checksum. The source ID of the ESP32-S2 should begin with `05E0` unless you changed it using the ESP-IDF sdkconfig.

fedechi, 2023-08-08:
no problem, have a good travel, we will go depth on all the issues in september. In the test there is only one device with esp_rdm lib and yes, i have modified UID from sdkconfig. Happy holidays.

someweisguy, 2023-08-21:
Issue #86 has been resolved, so I am hoping that the fixes in that issue have carried over to this issue. I will do some testing soon now that I have access to my ESP32-S2 again!

someweisguy, 2023-08-23:
This appears to be resolved on both ESP-IDF and Arduino. Let me know if you run into any more issues!

closed 2023-08-23 by someweisguy