someweisguy/esp_dmx #83: esp_dmx on ESP-IDF 5.1 not work?

habablaxe, 2023-07-27:
examples\idf_write
default setting
if i compile on  ESP-IDF 4.4.5 - all work! the device connected to the esp32 is receiving a DMX signal
if i compile on  ESP-IDF 5.1 - the device is not receiving a signal ((
any ideas?

habablaxe, 2023-07-27:
i downgraded the ESP-IDF version to:
5.0.3 - not work,
5.0.0 - not work,
4.4.4 - work!
Сan i get your sdkconfig file for version 5, maybe some default options are not enabled?

someweisguy, 2023-08-07:
Apologizes for the delay on this! I am currently moving across the country so my access to internet is spotty.

I am surprised to see that you're having issues with ESP-IDF v5.0.0. The latest tagged commit, v3.0.3-beta should work on IDF v5.0.0. 

Before I packed up my ESP32 hardware, I was testing the latest commit on branch `release/v3.1` on ESP-IDF v5.1. It appeared to be mostly working. Perhaps you can switch to that branch and try the latest version of ESP-IDF? The default sdkconfig should work.

habablaxe, 2023-08-18:
IDF v5.1
release/v3.1
idf_write.c
for me (I only need to write DMX) everything works, but there seems to be a problem when receive #86

closed 2023-08-18 by habablaxe