someweisguy/esp_dmx #85: General Question About Wireless Audio Transmitting

WeberIT, 2023-08-06:
Hi, i stumbled over your code while i was searching for a way to send audio data from a dj mixer with a 2 channel XLR / DMX output (1chan left, 1 chan right) to the amplifier - right now i am doing this by 2 long 3 pin dmx cables .

so i think it should be possible to send audio signal with your code or does it only work with dmx light packages ?

and the second question is would id be a solution to take a esp32 behind the dj mixer and plug the cable/cables from that into the esp32   (maybe 2 esp's dont know if  i can send 2 audio signals at the same time on 1 device ) 

and then at the amplifier the same but with dmx output where i can plug in the 2 cables coming from esp (amplifier) and connect them to the amp 

but it would be really nice if i could use 1 esp at each side to transmit and recieve the audio signal because the other way it would be 4 esp in this setup :( 

the nexst step would be to transfer the audio wireless from the amp to the speakers where each speaker has also 2 connectors so in the situation where i can only transfer 1 port at 1 esp only that would be plus 8 esp's for only 2 speakers ^^ and 10 in total with the audio source ;)

sorry for my bad english i hope you understand my question

someweisguy, 2023-08-07:
Thanks for reaching out! 

Unfortunately, this library is not able to send audio signals. It only sends DMX data, which is a digital signal that should not be plugged into audio amplifiers. It is possible for the ESP32 to send audio signals, but this library does not support this.

More information on DMX can be found [here](https://en.wikipedia.org/wiki/DMX512).

Hopefully this is helpful. Let me know if you have any other questions or concerns!

netmindz, 2023-08-18:
I have used a similar library to send audio data, but not the raw audio but the results of the FFT analysis used by WLED

https://github.com/netmindz/WLED-sync/blob/main/examples/E131DMX-WLED-Sync/E131DMX-WLED-Sync.ino

This sketch merges the dmx data over IP (e1.31/sACN) and the WLED Audio Sync data and puts out over wired DMX

someweisguy, 2023-08-18:
I'm closing this issue. If there are any more questions, please feel free to reopen! :)

closed 2023-08-18 by someweisguy