someweisguy/esp_dmx #87: Calling dmx_receive with wait_ticks=0 does never return data

arneboe, 2023-08-17:
From my understanding, if I call dmx_receive with `wait_ticks=0` I should receive data when data has been received since the last time I called it. But that never happens. It always returns 0. 
Is this a bug or is my understanding of the code incorrect? :)

netmindz, 2023-08-18:
See usage : https://github.com/MoonModules/WLED/pull/64

someweisguy, 2023-08-18:
Your understanding is correct! I will investigate further (after I resolve #86 :P).

someweisguy, 2023-08-19:
Now that #86 appears to be resolved, I've tested the `wait_ticks=0` behavior and it appears to be behaving as intended. 

Are you still seeing any issues with this?

arneboe, 2023-08-20:
I am no longer seeing the issue on the latest commit on` release/v3.1`
Thanks!

closed 2023-08-20 by arneboe