someweisguy/esp_dmx #104: Compile errors version 2 and up

MartijnCalysto, 2023-10-05:
I probably do something wrong, but it works with version 1.1.3. I get errors like: "'DMX_PACKET_SIZE' was not declared in this scope"
I use Arduino IDE 2.2, installed ESP32 Wroom dev board. I like to compile ArtNet to DMX Gateway, but I've tried examples too. Booth give me this kind of errors...

Many thanks for making this library!!

someweisguy, 2023-10-07:
Thank you for the kind words!

Is there a particular reason you are using esp_dmx v1.1.3? If not, I would recommend that you update to the latest version, v3.1.0. That should make the errors disappear.

MartijnCalysto, 2023-10-08:
I did, but it's the other way around, version 1.1.3 works, higher versions don't...

someweisguy, 2023-10-08:
> I use Arduino IDE 2.2, installed ESP32 Wroom dev board.

> I did, but it's the other way around, version 1.1.3 works, higher versions don't...

I see. Can you confirm which version of Arduino-ESP32 you are using? You should be using v2.0.3 or higher. I see that you are using IDE v2.2, but I'm specifically referring to the Arduino-ESP32 framework, not the IDE. You can verify this by checking the Boards Manager on the Arduino IDE and searching `esp32` to find "esp32 by Espressif Systems." I am currently using v2.0.14 without any issues.

MartijnCalysto, 2023-10-09:
I've version 2.0.14 too...  I've used this link in the Additional Board Manager URLs field: https://espressif.github.io/arduino-esp32/package_esp32_index.json

MartijnCalysto, 2023-10-09:
`C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:121:14: error: 'DMX_MAX_PACKET_SIZE' was not declared in this scope
 byte dataOut[DMX_MAX_PACKET_SIZE];
              ^~~~~~~~~~~~~~~~~~~
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:121:14: note: suggested alternative: 'ETH_MAX_PACKET_SIZE'
 byte dataOut[DMX_MAX_PACKET_SIZE];
              ^~~~~~~~~~~~~~~~~~~
              ETH_MAX_PACKET_SIZE
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:122:13: error: 'DMX_MAX_PACKET_SIZE' was not declared in this scope
 byte dataIn[DMX_MAX_PACKET_SIZE];
             ^~~~~~~~~~~~~~~~~~~
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:122:13: note: suggested alternative: 'ETH_MAX_PACKET_SIZE'
 byte dataIn[DMX_MAX_PACKET_SIZE];
             ^~~~~~~~~~~~~~~~~~~
             ETH_MAX_PACKET_SIZE
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino: In function 'void initDMXIn()':
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:302:30: error: 'DMX_DEFAULT_CONFIG' was not declared in this scope
   dmx_config_t dmxInConfig = DMX_DEFAULT_CONFIG;
                              ^~~~~~~~~~~~~~~~~~
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:302:30: note: suggested alternative: 'ETH_DEFAULT_CONFIG'
   dmx_config_t dmxInConfig = DMX_DEFAULT_CONFIG;
                              ^~~~~~~~~~~~~~~~~~
                              ETH_DEFAULT_CONFIG
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:303:3: error: 'dmx_param_config' was not declared in this scope
   dmx_param_config(dmxInPort, &dmxInConfig);
   ^~~~~~~~~~~~~~~~
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:303:3: note: suggested alternative: 'dmxInConfig'
   dmx_param_config(dmxInPort, &dmxInConfig);
   ^~~~~~~~~~~~~~~~
   dmxInConfig
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:314:33: error: 'DMX_MAX_PACKET_SIZE' was not declared in this scope
   dmx_driver_install(dmxInPort, DMX_MAX_PACKET_SIZE, queueSize, &queue, interruptPriority);
                                 ^~~~~~~~~~~~~~~~~~~
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:314:33: note: suggested alternative: 'ETH_MAX_PACKET_SIZE'
   dmx_driver_install(dmxInPort, DMX_MAX_PACKET_SIZE, queueSize, &queue, interruptPriority);
                                 ^~~~~~~~~~~~~~~~~~~
                                 ETH_MAX_PACKET_SIZE
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino: In function 'void initDMXOut()':
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:320:31: error: 'DMX_DEFAULT_CONFIG' was not declared in this scope
   dmx_config_t dmxOutConfig = DMX_DEFAULT_CONFIG;
                               ^~~~~~~~~~~~~~~~~~
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:320:31: note: suggested alternative: 'ETH_DEFAULT_CONFIG'
   dmx_config_t dmxOutConfig = DMX_DEFAULT_CONFIG;
                               ^~~~~~~~~~~~~~~~~~
                               ETH_DEFAULT_CONFIG
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:321:3: error: 'dmx_param_config' was not declared in this scope
   dmx_param_config(dmxOutPort, &dmxOutConfig);
   ^~~~~~~~~~~~~~~~
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:321:3: note: suggested alternative: 'dmxOutConfig'
   dmx_param_config(dmxOutPort, &dmxOutConfig);
   ^~~~~~~~~~~~~~~~
   dmxOutConfig
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:332:34: error: 'DMX_MAX_PACKET_SIZE' was not declared in this scope
   dmx_driver_install(dmxOutPort, DMX_MAX_PACKET_SIZE, queueSize, NULL, interruptPriority);
                                  ^~~~~~~~~~~~~~~~~~~
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:332:34: note: suggested alternative: 'ETH_MAX_PACKET_SIZE'
   dmx_driver_install(dmxOutPort, DMX_MAX_PACKET_SIZE, queueSize, NULL, interruptPriority);
                                  ^~~~~~~~~~~~~~~~~~~
                                  ETH_MAX_PACKET_SIZE
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:337:28: error: 'DMX_MODE_TX' was not declared in this scope
   dmx_set_mode(dmxOutPort, DMX_MODE_TX);
                            ^~~~~~~~~~~
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:337:28: note: suggested alternative: 'WIFI_MODE_MAX'
   dmx_set_mode(dmxOutPort, DMX_MODE_TX);
                            ^~~~~~~~~~~
                            WIFI_MODE_MAX
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:337:3: error: 'dmx_set_mode' was not declared in this scope
[... cut ...]
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:399:9: error: 'dmx_wait_tx_done' was not declared in this scope
         dmx_wait_tx_done(dmxOutPort, DMX_TX_PACKET_TOUT_TICK);
         ^~~~~~~~~~~~~~~~
C:\Users\marti\OneDrive\Documenten\Arduino\projects\ESP32_ArtNet_DMX_Gateway\ESP32_ArtNet_DMX_Gateway.ino:399:9: note: suggested alternative: 'dmx_wait_sent'
         dmx_wait_tx_done(dmxOutPort, DMX_TX_PACKET_TOUT_TICK);
         ^~~~~~~~~~~~~~~~
         dmx_wait_sent

exit status 1

Compilation error: 'DMX_MAX_PACKET_SIZE' was not declared in this scope`

someweisguy, 2023-10-10:
Oh, I think I see the issue. The constants in that error message are no longer valid. I think somehow the examples you are using are the examples for the v1 or v2 library. v3 is not compatible with v1 or v2. 

[Here](https://github.com/someweisguy/esp_dmx/blob/v3.1.0/examples/DMXWrite/DMXWrite.ino) is the link to the v3 DMX Write example. I just tested this code on Arduino and it compiled for me.

MartijnCalysto, 2023-10-10:
Don't know why but this was it!! Yes!! Do you know a project like ConnoDMX that's works with eps_dmx v3?

someweisguy, 2023-10-10:
That's great to hear!

I just released v3, so I am not aware of any other projects currently using it.

closed 2023-10-12 by MartijnCalysto