someweisguy/esp_dmx #108: Troubles with ETC Express

kpelzel, 2023-10-28:
First off, thank you so much for this library and the documentation, it's incredibly valuable. 

I'm struggling to get this to work properly with an old ETC Express. I'm running the DMX Read example (with minor tweaks to timing) just trying to read a few channels within the dmx packet, but the packet size is never correct. I'm pretty sure the ETC Express will always send the full 513 bytes for every packet. I may have done something wrong, but it looks like the example just chunks up the dmx packet randomly. I swear there was a point where I was getting consistent packet sizes of 513, but I'm not seeing that anymore. Any help is greatly appreciated.

## Setup:
- ETC Express (v3.1). [dmx timings](https://support.etcconnect.com/ETC/FAQ/DMX_Speed)
- [sparkfun dmx shield](https://www.sparkfun.com/products/15110)
- esp32 do it devkit v1

## Code: 
```c
#include "esp_dmx.h"
#include "esp_log.h"
#include "esp_system.h"
#include "freertos/task.h"

#define TX_PIN 17 // The DMX transmit pin.
#define RX_PIN 16 // The DMX receive pin.
#define EN_PIN 15 // The DMX transmit enable pin.

static const char *TAG = "main"; // The log tagline.

static uint8_t data[DMX_PACKET_SIZE] = {}; // Buffer to store DMX data

void app_main()
{
  const dmx_port_t dmx_num = DMX_NUM_1;

  // Set communication pins and install the driver
  dmx_config_t config = DMX_CONFIG_DEFAULT;
  dmx_driver_install(dmx_num, &config, DMX_INTR_FLAGS_DEFAULT);
  dmx_set_pin(dmx_num, TX_PIN, RX_PIN, EN_PIN);
  dmx_set_baud_rate(dmx_num, DMX_BAUD_RATE); // Set DMX baud rate.
  dmx_set_break_len(dmx_num, 94);            // Set DMX break length.
  dmx_set_mab_len(dmx_num, 26);              // Set DMX MAB length.

  dmx_packet_t packet;
  bool is_connected = false;

  TickType_t last_update = xTaskGetTickCount();
  while (true)
  {
    // Block until a packet is received
    if (dmx_receive(dmx_num, &packet, DMX_TIMEOUT_TICK))
    {
      const TickType_t now = xTaskGetTickCount();

      if (!is_connected)
      {
        // Log when we first connect
        ESP_LOGI(TAG, "DMX is connected.");
        is_connected = true;
      }

      // if (now - last_update >= pdMS_TO_TICKS(1000))
      {
        // Only read data every 1000ms
        dmx_read(dmx_num, data, DMX_PACKET_SIZE);
        ESP_LOGI(TAG, "error: %i", packet.err);
        ESP_LOGI(TAG, "Start code: %02x, Size: %i", packet.sc, packet.size);
        ESP_LOG_BUFFER_HEX(TAG, data, 16); // Log first 16 bytes
        last_update = now;
      }
    }
    else if (is_connected)
    {
      // DMX timed out after having been previously connected
      ESP_LOGI(TAG, "DMX was disconnected.");
      // break;
    }
[... 144 lines cut ...]
[0;32mI (73095) main: error: 0[0m
[0;32mI (73105) main: Start code: 00, Size: 2[0m
[0;32mI (73105) main: 00 00 00 00 00 00 00 01 00 00 00 01 00 28 d5 9f [0m
[0;32mI (73115) main: error: 0[0m
[0;32mI (73115) main: Start code: c0, Size: 62[0m
[0;32mI (73115) main: c0 ff b4 b2 d5 49 71 98 75 ee f3 ac 51 d1 eb b4 [0m
[0;32mI (73125) main: error: 0[0m
[0;32mI (73125) main: Start code: fc, Size: 1[0m
[0;32mI (73135) main: fc 00 00 01 01 00 01 00 00 00 e5 ff ba 43 a8 bf [0m
[0;32mI (73145) main: DMX was disconnected.[0m
[0;32mI (73145) main: error: 0[0m
[0;32mI (73145) main: Start code: 01, Size: 69[0m
[0;32mI (73155) main: 01 00 22 61 23 27 cc 02 c5 4f a1 71 06 c3 fc fb [0m
[0;32mI (73155) main: DMX was disconnected.[0m
[0;32mI (73165) main: error: 0[0m
[0;32mI (73165) main: Start code: 01, Size: 1[0m
[0;32mI (73175) main: 01 00 ff 00 00 00 00 6c e1 b1 67 df 41 33 02 2c [0m
[0;32mI (73175) main: error: 0[0m
[0;32mI (73185) main: Start code: 00, Size: 1[0m
[0;32mI (73185) main: 00 00 00 00 00 08 01 00 00 00 67 df 41 33 02 2c [0m
[0;32mI (73195) main: DMX was disconnected.[0m
[0;32mI (73195) main: error: 0[0m
[0;32mI (73195) main: Start code: e2, Size: 4[0m
[0;32mI (73205) main: e2 13 ef c4 00 00 e0 00 00 01 01 01 00 fa 20 30 [0m
```

kpelzel, 2023-10-28:
## Further Findings
I think I was wrong about the etc express sending all 513 bytes. It appears it may only be sending 96 bytes. I was also able to get some full packets so I know it's working to some extend, but there are a lot of smaller broken packets between the full packets. When I ignore all packets that aren't a length of 513 I repeated (but very slowly) get this output:
```
[0;32mI (395376) main: Start code: 00, Error: 0, Size: 513[0m
[0;32mI (395376) main: 00 73 ff ff ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395376) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395386) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395396) main: 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 [0m
[0;32mI (395396) main: 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 [0m
[0;32mI (395406) main: 00 00 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395416) main: 00 73 ff ff ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395416) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395426) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395436) main: 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 [0m
[0;32mI (395446) main: 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 [0m
[0;32mI (395446) main: 00 00 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395456) main: 00 73 ff ff ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395466) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395466) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395476) main: 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 [0m
[0;32mI (395486) main: 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 [0m
[0;32mI (395486) main: 00 00 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395496) main: 00 73 ff ff ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395506) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395506) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395516) main: 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 [0m
[0;32mI (395526) main: 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 [0m
[0;32mI (395526) main: 00 00 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395536) main: 00 40 ff ff ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395546) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395546) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395556) main: 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 [0m
[0;32mI (395566) main: 00 00 00 00 00 00 00 00 00 f9 ff 00 00 00 00 00 [0m
[0;32mI (395566) main: 00 00 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395576) main: 00 73 ff ff ff 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395586) main: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [0m
[0;32mI (395586) main: 00 [0m
```

I tried setting my packet size to 96 and also 97, but it didn't improve the performance

someweisguy, 2023-10-29:
Thanks for the information and the kind words!

This is a peculiar issue - it appears that the ESP32 is having a hard time determining when the DMX packet starts and ends, which is why you are seeing shifted bytes. It's surprising because of how robust the ETC Express is.

I don't have access to an Express to test, but I have loaded up esp_dmx v3.1.0 on my ESP32 and connected to my Entec DMX dongle and it is able to read DMX properly using your code. I also attempted to emulate an Express by sending DMX from a spare ESP32 using the DMX timings that the Express uses.

Which version of ESP-IDF are you using? And are you able to receive DMX coming from a different console?

kpelzel, 2023-10-30:
Unfortunately I don't have another console to test. I'm running ESP-IDF v5.1.1. I'll try out a different ESP32 and maybe a different power source. Currently I've been running off usb power from my laptop for testing, but I assumed that'd be enough for the esp32 and dmx shield. 

My only other guess is incorrect wiring between the esp32 and dmx-shield, but I assume I wouldn't be seeing any packets if that was the case.

kpelzel, 2023-10-31:
I checked the signal with an oscilloscope today to make sure I'm not going crazy. It looks like the timings are fine. 
Interpacket idle = ~11ms
Break = 94μs
MAB = 27μs

One question though. Is my signal inverted? Every diagram of dmx I've seen shows the idle low, break high, and MAB low. Mine seems to be the opposite. Are my D+ and D- getting crossed somewhere?

[image]
[image]
[image]

someweisguy, 2023-11-07:
If you've connected your scope to the `A` line of your DMX bus, then your idle should be high; break, low; and MAB, high.

If you haven't tried already, one thing that I always do when troubleshooting is to connect my scope directly to the UART pins on the ESP32 to ensure that the DMX signal is making it from the RS485 transceiver to the ESP32. It is possible that the enable pin is misconfigured which would make it impossible for the ESP32 to read data.

kpelzel, 2023-11-14:
Alright well, I'm not exactly sure what I did, but it's working perfectly now. I'm gonna assume it was bad wiring somewhere or theater ghosts. Thanks for all the help. Just some notes:

1. That scope reading is from the rx output after the max485, not straight from the dmx. 
2. I was wrong about the etc express only sending 96 channels, turns out the patch on the board was wrapping around after 96 and causing the packet to have a pattern of repeated values. 

Here's the code for the curious: https://github.com/kpelzel/LALT-dmx

closed 2023-11-14 by kpelzel