someweisguy/esp_dmx #120: rdm_register_dmx_personality missing in 3.1.0

netmindz, 2024-01-15:
The rdm_register_dmx_personality method appears to have disappeared when I tried to update to 3.1.0

someweisguy, 2024-01-17:
Thanks for submitting this. I apologize in advance because it's been a while since I've looked at v3.1.0 and earlier. Were you able to call `rdm_register_dmx_personality()` in any earlier versions? I don't recall implementing this method in v3.

This method will be implemented in the v4.0.0

netmindz, 2024-01-17:
I was just updating a PR for an update to WLED-MM and saw he had swapped the usage of 3.0-beta to a specific hash, saw you had a release of RDM as 3.1.0 and wrongly assumed that would be the right version, I didn't spot you've moved onto 4.x

closed 2024-01-17 by netmindz

netmindz, 2024-01-17:
https://github.com/someweisguy/esp_dmx/commit/c6563ba3212cbbe02095fe38ccadf567f5ba7550

Looks like it will be in the 3.1.1 release, when that happens

someweisguy, 2024-02-02:
v4.0.0 has been published. This fixes this issue. Thanks for your patience!