someweisguy/esp_dmx #121: DMX Read - Unstable packet.size depends on the number of DMX channels

Kokichiro, 2024-01-16:
### Describe the bug
If the number of received DMX channels is certain amount near 512ch, packet.size will be unstable.

### To reproduce

- Board: M5Stack CoreS3 - a kind of ESP32-S3
- IDE: Arduino IDE (with M5Unified.hpp installed)
- esp_dmx: v3.1.0
- Use DMXRead.ino sample, but just add Serial.printf("packet.size is %d and DMX_PACKET_SIZE is %d\n", packet.size, DMX_PACKET_SIZE); after Serial.printf("Start code is 0x%02X and slot 1 is 0x%02X\n", data[0], data[1]); 
- Input DMX Signal at various number of channels

### Result
When the number of channel is 512, and slot 1 is fixed 0x97,
Start code is 0x00 and slot 1 is 0x97
packet.size is 52 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 396 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 142 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 487 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 257 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 27 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 370 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 192 and DMX_PACKET_SIZE is 513
.....
(sometimes Serial.println("DMX was disconnected."); is called and DMX input stops.)

When the number of channel is 512, and slot 1 is fixed 0x97,
Start code is 0x00 and slot 1 is 0x00
packet.size is 511 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x00
packet.size is 511 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x00
packet.size is 511 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x00
packet.size is 511 and DMX_PACKET_SIZE is 513
....

When the number of channel is 510, and slot 1 is fixed 0x97,
Start code is 0x00 and slot 1 is 0x97
packet.size is 511 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 511 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 511 and DMX_PACKET_SIZE is 513
...

When the number of channel is 500, and slot 1 is fixed 0x97,
Start code is 0x00 and slot 1 is 0x97
packet.size is 501 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 501 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 501 and DMX_PACKET_SIZE is 513
Start code is 0x00 and slot 1 is 0x97
packet.size is 501 and DMX_PACKET_SIZE is 513

### Analysys
From above result, it can be said
- 512ch input causes unstable packet.size, also unstable DMX input
- 511ch input causes wrong reading of Slot 1 (it should be 0x97, but forced to 0x00)
- 510ch or less input causes stable DMX input

I don't know if this is specific to M5Core or general ESP32-S3 issue.

### Environment
OS: Windows 10
IDE &IDE Version: Arduino IDE 2.2.1
Repository Version: 3.1.0 (the newest one)

someweisguy, 2024-01-17:
Thanks for the report. I am going to investigate this issue. 

Using the base ESP32, I am unable to replicate this issue exactly. But I am seeing that when the ESP32 receives 512 slots, it reports the `packet.size` as 511. It reports this value consistently. I suspect these issue are related.

Kokichiro, 2024-01-17:
For temporary solution, if there is a way to "regulate" the input channel count, that will work for the people who work with small shows and don't need full 512ch capability.

someweisguy, 2024-02-02:
I believe I have fixed this issue in v4.0.0. I ran some ad-hoc tests that passed. I'll close this for now; let me know if you have any questions!

closed 2024-02-02 by someweisguy