someweisguy/esp_dmx #124: compiling error with 3.10 but not with 3.0.3-beta

doncogit, 2024-01-29:
hi,

i used 3.0.3 beta before and it worked fine with arduino ide 2.2.1 and esp32-s3 chip.
now with 3.10 i get errors and it doesnt compile. When i switch back to 3.03. beta same code works again:

`\Documents\Arduino\v14\v14.ino: In function 'void setup()':
Documents\Arduino\v14\v14.ino:154:33: error: 'DMX_DEFAULT_INTR_FLAGS' was not declared in this scope
     dmx_driver_install(dmx_num, DMX_DEFAULT_INTR_FLAGS);
                                 ^~~~~~~~~~~~~~~~~~~~~~
\Documents\Arduino\v14\v14.ino:154:33: note: suggested alternative: 'IP_DEFAULT_TTL'
     dmx_driver_install(dmx_num, DMX_DEFAULT_INTR_FLAGS);
                                 ^~~~~~~~~~~~~~~~~~~~~~
                                 IP_DEFAULT_TTL
\Documents\Arduino\v14\v14.ino: In function 'void dmxsetup()':
\Documents\Arduino\v14\v14.ino:639:31: error: 'DMX_DEFAULT_INTR_FLAGS' was not declared in this scope
   dmx_driver_install(dmx_num, DMX_DEFAULT_INTR_FLAGS);
                               ^~~~~~~~~~~~~~~~~~~~~~
\Documents\Arduino\v14\v14.ino:639:31: note: suggested alternative: 'IP_DEFAULT_TTL'
   dmx_driver_install(dmx_num, DMX_DEFAULT_INTR_FLAGS);
                               ^~~~~~~~~~~~~~~~~~~~~~
                               IP_DEFAULT_TTL
Multiple libraries were found for "SD.h"
  Used: \AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\SD
  Not used: \AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: 'DMX_DEFAULT_INTR_FLAGS' was not declared in this scope`

someweisguy, 2024-01-30:
Hi, thanks for reaching out. Try `DMX_INTR_FLAGS_DEFAULT`. Let me know if you have any more issues! :)

doncogit, 2024-01-30:
thansk, but still gives me errors, but now others:

`Compilation error: too few arguments to function 'bool dmx_driver_install(dmx_port_t, const dmx_config_t*, int)'`

someweisguy, 2024-02-02:
Glad you got it figured out! Speaking of change, I just published v4.0.0! You're welcome to check that out or stay on v3.1.

For context, v3 was my first try at making RDM work so I was changing code constantly. My main goal with v4 is to try to stay on v4 for as long as possible and only make minor, non-breaking changes. I spent a lot of time looking at weird edge cases in RDM and figuring out how to handle them. I believe I have a path forward figured out to continue supporting RDM without making any breaking changes in the near future! Thanks for your patience :)

In the meantime, I will close this issue. Let me know if you have any other questions!

closed 2024-02-02 by someweisguy