someweisguy/esp_dmx #128: Bug string constant to 'char*' [-Wwrite-strings]

bypepe, 2024-02-21:
ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

/** @brief The default configuration for the DMX driver.*/
#define DMX_CONFIG_DEFAULT                                            \
  (dmx_config_t) {                                                    \
    DMX_INTR_FLAGS_DEFAULT,           /*interrupt_flags*/             \
        32,                           /*root_device_parameter_count*/ \
        0,                            /*sub_device_parameter_count*/  \
        0,                            /*model_id*/                    \
        RDM_PRODUCT_CATEGORY_FIXTURE, /*product_category*/            \
        ESP_DMX_VERSION_ID,           /*software_version_id*/         \
        ESP_DMX_VERSION_LABEL,        /*software_version_label*/      \
        32,                           /*queue_size_max*/              \
  }

someweisguy, 2024-02-22:
Thanks for the find. I had been programming in C and forgot to check against C++ standards. :)

I have this fixed and will be verifying the fix before closing this issue.

someweisguy, 2024-02-22:
Fixed. This will be included in the next patch.

closed 2024-02-22 by someweisguy