someweisguy/esp_dmx #143: ESP32-S3 RDM issues with UART0

lofty84, 2024-02-28:
Hi, firstly thankyou for creating such an in depth library!
I am encountering an issue with RDM on ESP32/S3 if you could possibly assist?
Using UART0 (unfortunately the only one I have available on the project) the device is not discoverable when loaded with the RDM Responder example. Using UART1 it is discoverable.
DMX receiving without RDM works absolutely fine on both UART0 and UART1.
I can see the RTS pin going high when an RDM discovery is performed and looks like some data is coming out of the UART0 Tx pin responding to the discovery, but unfortunately it is not recognised by the discoverer. UART1 seems to have no issues.
Do you have any advice? Other than to not use UART0 🙂

lofty84, 2024-02-28:
Platform is Arduino and USB CDC is disabled btw

someweisguy, 2024-02-29:
Thanks for the kind words!

This is a great question. It is possible that there is a hardware limitation with UART0 on the S3. I may be able to patch it if this is the case. I will investigate!

someweisguy, 2024-03-01:
Are you comfortable editing library code? If so, I'd like you to try something for me. I think it would be faster than me trying to determine how to test this.

I think I may have an if-statement in the UART initialization code that is causing the problem for you. It can be found [here](https://github.com/someweisguy/esp_dmx/blob/v4.0.1/src/dmx/hal/uart.c#L328-L337). Currently, the code looks like this:

```c
  if (dmx_num != 0) {  // Default UART port for console
#if SOC_UART_REQUIRE_CORE_RESET
    // ESP32C3 workaround to prevent UART outputting garbage data
    uart_ll_set_reset_core(uart->dev, true);
    periph_module_reset(uart_periph_signal[dmx_num].module);
    uart_ll_set_reset_core(uart->dev, false);
#else
    periph_module_reset(uart_periph_signal[dmx_num].module);
#endif
  }
```

I believe the `if (dmx_num != 0)` needs to be edited. If you are comfortable doing so, try changing it to `if (true)` so that the if-statement is always executed. I think this may solve your problem until I am able to patch this.

lofty84, 2024-03-01:
Yes absolutely, I’m more than happy to modify the library and test it. It’ll be a couple of days before I can undertake testing but as soon as I do I’ll let you know how it goes.
Many thanks!

lofty84, 2024-03-05:
I've managed to test this today but it still has the same issue I'm afraid.
I assume its related to the ESP32-S3 having on-board USB serial debug for USB attached to this UART (even though I have disabled it) so I doubt there is much else that can be done unfortunately.
Many thanks for looking into this though!

closed 2024-03-05 by lofty84