someweisguy/esp_dmx #146: Start Address Not Saving Via RDM

lofty84, 2024-03-13:
Board: ESP32-S3

Platform: Arduino 

Issue: It is possible to set the start address of a device by RDM and this address is stored fine during operation. However upon powering down the device and powering up again the address is lost and the device defaults to address 1.
It seems like the start address is not being saved to the flash.

lofty84, 2024-03-14:
Also tested on 2 different ESP32-S3 boards including using the RDM Responder example with the same result

someweisguy, 2024-03-15:
Thanks for the report - I was able to reproduce this issue.

I am not too surprised that this issue occurred - I apologize for the inconvenience. The reason this bug is happening is because I was attempting to fix a weird problem. The problem is that, in some configurations of this library, saving parameters such as the DMX start address to flash prevents the ESP32 from reading DMX. To be more specific, when the DMX driver is not stored in the ESP32 IRAM, this error can occur. To prevent a potential issue wherein the ESP32 fails to read an RDM message because it is writing to flash, this library will only write to flash when:

1. A parameter (like DMX start address) has been updated and needs to be written to flash
2. When a non-RDM packet has been received.

Clearly, I need to take another pass at trying to solve this problem. I'll investigate and see what I can come up with. Thanks for your patience!

lft8, 2024-03-15:
no need to apologise, you're doing great work on this library :)
if there's anything you'd like me to test then let me know

lft8, 2024-03-20:
I've been thinking about this and personally I can't see why temporarily stopping receiving DMX during saving the start address to flash would be an issue. Setting the start address of a device would be done during commissioning of an installation and not in routine operation, I've had a look at a commercial RDM fixture and I do believe it is unresponsive for a very short period after setting the start address by RDM

someweisguy, 2024-03-21:
I think you may be right here. I pushed a potential fix to the branch `fix/rdm-eeprom`. It saves data to flash after receiving any DMX or RDM packet. Give it a try, if you are able. I am going to do some testing as well.

lft8, 2024-03-22:
I have tested quite thoroughly and no reported issues :)
I'll close the issue

closed 2024-03-22 by lft8