someweisguy/esp_dmx #150: Problem with installing DMX on UART2

k-dedman, 2024-03-27:
Any time I try to set up UART2 with dmx_driver_install the program crashes. I am trying to setup all 3 UART ports with a DMX output, I do not need RX or the enable pin. I have tried setting up UART2 on an ESP32-WROOM-32D and a WT32-ETH01 module, both modules crash when trying to setup UART2. UART0 & UART1 are working as intended. I have stripped my project to only setup UART2 and it still crashes. I am using Arduino IDE with esp_dmx 4.1.0

`
dmx_port_t dmxPort0 = DMX_NUM_0;
dmx_port_t dmxPort1 = DMX_NUM_1;
dmx_port_t dmxPort2 = DMX_NUM_2;

uint8_t dmx0_tx_pin = 1;
uint8_t dmx1_tx_pin = 18;
uint8_t dmx2_tx_pin = 21;

void dmx_setup(void)
{
    dmx_config_t config0 = DMX_CONFIG_DEFAULT;
    dmx_config_t config1 = DMX_CONFIG_DEFAULT;
    dmx_config_t config2 = DMX_CONFIG_DEFAULT;
    dmx_personality_t personalities0[] = {};
    dmx_personality_t personalities1[] = {};
    dmx_personality_t personalities2[] = {};
    int personality_count0 = 0;
    int personality_count1 = 0;
    int personality_count2 = 0;
    dmx_driver_install(dmxPort0, &config0, personalities0, personality_count0);
    dmx_set_pin(dmxPort0, dmx0_tx_pin, -1, -1);
    dmx_driver_install(dmxPort1, &config1, personalities1, personality_count1);
    dmx_set_pin(dmxPort1, dmx1_tx_pin, -1, -1);
    dmx_driver_install(dmxPort2, &config2, personalities2, personality_count2); // Crashes here
    dmx_set_pin(dmxPort2, dmx2_tx_pin, -1, -1);

}
`

riwalker, 2024-03-27:
assume you checked the pins, some UARTS have non moveable UART functions (i.e the C6 UART2 has to be on 5 & 4)
I submitted some changed which were included in 4.1.0 for UART2 specifically for the LP (Low-Power) UART which is UART2 on the C6 family.  it may be these changes which are now jarring with the older (none LP) UARTS.
look at uart.c, we may need some additional changes to check for non LP uarts, or only apply the recent changes ONLY for the C6

k-dedman, 2024-03-27:
Thanks for the reply.

The problem is the code never gets to the point where it can set the pins for UART2, if i comment everything out but the dmx_driver_install for UART2, it still crashes. 

I looked into the CONFIG_IDF_TARGET_ESP32C6 #ifdef's and commented out the code just to make sure it was initialising UART2 normally and it still crashes. The pins for UART2 should be remappable for the WT32-ETH01 dev board.

I have also made sure that I can setup a serial output on UART2 and that works, so the port can definitely be used. 

I have got DMX output working for UART0 (tx pin 1), UART1 (tx pin 17 or 14), but I cannot get it to work on UART2

k-dedman, 2024-03-28:
Ok I got it working.

Not sure whats going wrong but heres how I fixed it.

The program was failing whenever it hit uart_ll_set_sclk due to a null pointer at uart->dev
I removed the check for including UART2 in the dmx_uart_t struct initialisation and the dmx port constant enum. I also explicitly defined &UART2 instead of using the UART_LL_GET_HW function.
```
static struct dmx_uart_t {
  const int num;
  uart_dev_t *const dev;
  intr_handle_t isr_handle;
} dmx_uart_context[DMX_NUM_MAX] = {
    {.num = 0, .dev = UART_LL_GET_HW(0)},
    {.num = 1, .dev = UART_LL_GET_HW(1)},
// #if DMX_NUM_MAX > 2
    {.num = 2, .dev = &UART2},
// #endif
};
```
```
/** @brief DMX port constants.*/
enum {
  DMX_NUM_0, /** @brief DMX port 0.*/
  DMX_NUM_1, /** @brief DMX port 1.*/
// #if SOC_UART_NUM > 2
  DMX_NUM_2, /** @brief DMX port 2.*/
// #endif
  DMX_NUM_MAX /** @brief DMX port max. Used for error checking.*/
};
```
I now have 3 seperate DMX TX outputs tested and working. Im not sure why this fix works as the ESP32 core should have 3 UARTS available and I had tested this with HardwareSerial.

riwalker, 2024-03-28:
Agree, I also in past versions had to code out the same struct #if test for UART2 and force it, although the latest code works for the LP UART on the C6.
its been a nightmare for the latest esp-idf code rolls, they are making huge code changes around the LP UART, hence why I provided the new routines for UART2 init, its changed every time for the last 3 or 4 esp-idf library release (5.0 to 5.2+)

GIPdA, 2024-04-11:
Hello!
Got the same issue today with ESP-IDF and the v4.1.0 release, for an ESP32-C3.
I fixed it with a small change in uart.c, line 32:

```
#if SOC_UART_NUM > 2
```

Currently ```DMX_NUM_MAX``` is used, but it's an enum and cannot be used in pre-processor.

bensuffolk, 2024-06-11:
This sounds like a problem I have having as well. I have just tried to upgrade from 3.0.3-beta that I have been successfully using on esp-idf 4.4.5 with DMX_NUM_1 and DMX_NUM_2 as the 2 UARTS.

I am using an original ESP32-WROVER

Now I have 4.1.0 it crashes when trying to install the dmx driver for UART2. I have not changed the version of esp-idf.

```  
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

...

#0  0x400fb617:0x3ffb8240 in uart_ll_set_sclk at /Users/suffo_b/.platformio/packages/framework-espidf/components/hal/esp32/include/hal/uart_ll.h:76
      (inlined by) dmx_uart_init at components/esp_dmx-4.1.0/src/dmx/hal/uart.c:357
#1  0x400f779d:0x3ffb8260 in dmx_driver_install at components/esp_dmx-4.1.0/src/dmx/driver.c:226
```

riwalker, 2024-06-11:
Seems the same as issue #150

https://github.com/someweisguy/esp_dmx/issues/150#issuecomment-2050224824

bensuffolk, 2024-06-12:
> Hello! Got the same issue today with ESP-IDF and the v4.1.0 release, for an ESP32-C3. I fixed it with a small change in uart.c, line 32:
> 
> ```
> #if SOC_UART_NUM > 2
> ```
> 
> Currently `DMX_NUM_MAX` is used, but it's an enum and cannot be used in pre-processor.

Can confirm this is the correct solution and is working well. I have created a pull request for #158 for when @someweisguy has an opportunity to merge it.

davidlmorris, 2024-08-10:
I can confirm this issue.  Simply take the Arduino_DMXWrite.ino file example (I'm using PlatformIO so I've changed it to Arduino_DMXWrite.cpp).

Is there any news on when we get a fix?  

And can anyone tell me when was the last 'good' version for using DMX_NUM_2?  (I used 2.0.2, but I see the timing and mark information have improved hugely since then, and I would be keen to use the latest).

It works on when `dmx_port_t dmxPort = 1` and crashes with `dmx_port_t dmxPort = 2` [Line 32](https://github.com/someweisguy/esp_dmx/blob/release/v4.1/examples/Arduino_DMXWrite/Arduino_DMXWrite.ino).

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2   
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028 
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d2157  PS      : 0x00060c30  A0      : 0x800d1d1d  A1      : 0x3ffb21c0  
A2      : 0x00000002  A3      : 0x3ffb8a18  A4      : 0x00000400  A5      : 0x00000004  
A6      : 0x3f406d20  A7      : 0x3ffbdb94  A8      : 0x00000000  A9      : 0x08000000  
A10     : 0x00800000  A11     : 0x3ff000c4  A12     : 0x3ffbdb7c  A13     : 0x3ffbdc2c  
A14     : 0x3ff000c4  A15     : 0x06ff1ff8  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000020  LBEG    : 0x40086c71  LEND    : 0x40086c93  LCOUNT  : 0x00000000  

Backtrace: 0x400d2154:0x3ffb21c0 0x400d1d1a:0x3ffb21f0 0x400d181d:0x3ffb2240 0x400d6f9e:0x3ffb2290
  #0  0x400d2154 in uart_ll_set_sclk at C:/Users/David/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/hal/esp32/include/hal/uart_ll.h:76    
      (inlined by) dmx_uart_init at .pio/libdeps/esp32doit-devkit-v1/esp_dmx/src/dmx/hal/uart.c:357
  #1  0x400d1d1a in dmx_driver_install at .pio/libdeps/esp32doit-devkit-v1/esp_dmx/src/dmx/driver.c:226
  #2  0x400d181d in setup() at src/Arduino_DMXWrite.cpp:58
  #3  0x400d6f9e in loopTask(void*) at C:/Users/David/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

Note line 58 is: ```  dmx_driver_install(dmxPort, &config, personalities, personality_count);```.

Line 226 in esp_dmx/src/dmx/driver.c
```
 // Initialize the UART peripheral
  if (!dmx_uart_init(dmx_num, driver, interrupt_flags)) {
    dmx_driver_delete(dmx_num);
    DMX_CHECK(false, false, "UART init error");
  }
```
Line 357 esp_dmx/src/dmx/hal/uart.c
```
  uart_ll_set_sclk(uart->dev, UART_SCLK_APB);
```

I am using IDF version 4.4.7 in the Arduino platform.

PlatormIO.ini relevant details:
```
[env:esp32doit-devkit-v1]
platform = espressif32 @ 6.7.0
board = esp32doit-devkit-v1
framework = arduino
build_flags = -ftrack-macro-expansion=0
	-fno-diagnostics-show-caret
lib_deps = someweisguy/esp_dmx
```

bensuffolk, 2024-08-11:
Did you download the latest version, this was fixed in a pull request #158 and is working fine for me now.

einlichtvogel, 2025-05-12:
I can confirm @k-dedman 's  solution for this problem!

Downloading the 4.1.0 Version from the Arduino Libary Manager does not work.
@someweisguy any updated from you here?

second-string, 2025-11-18:
Just an FYI, the fix was merged (obviously) to the `release/v4.1` branch, but the `v4.1.0` tag still points to the commit right before. If you have the time to update the tag quickly it would be appreciated please @someweisguy. Not a huge deal though