someweisguy/esp_dmx #153: ESP32 DMX Libary packet size read problem

nnovation, 2024-05-12:
esp dmx library very help full for me but i noted one issue please solve this is issues - issues when read dmx so packet size read but when run greater then more packet after run less packet size value of packet size before data tranfered packet size. for more understand give example - when i run 200 channel packet size so read 200 but after run more then 200 like a 250 but read packet size 200 if i send data less then 200 like 150 then read packet size so read 150 but after then mote then 150 but in actual shw 150 packet size. so not increass but decrease packet size. if i reset esp32 then ok to read more packet size. 

Thanks you

someweisguy, 2024-06-14:
Thanks for the question. Which version of this library are you using and what version of ESP32 are you using?

nnovation, 2024-06-14:
I used latest version 4.1.0 esp_dmx library and I used ESP32 Devkit module and esp32 board version 2.0.17.

2345lug, 2024-09-22:
Hello! I have a same problem with latest version.

nnovation, 2024-09-22:
Hello someweisguy. This is library for me very usefull. Please solve this issue.

putzlicht, 2025-01-25:
Still unsolved. Time to take a look on the math?

maxroseGIT, 2025-04-14:
Hi @putzlicht,

interesting insight and thanky for the interpretation of the documentation of _esp_dmx_.

Do you think that your findings play a role when using _esp_dmx_ with professional DMX gear like [Wolfmix W1](https://wolfmix.com/en/wolfmix-w1)?

I have some trouble getting all channels read on my ESP32 when it receives data from the Wolfmix DMX controller. I wonder if this controller sends different packet sizes to obtains its claimed performance? Which then in turn gives me the troubles to read the packets correctly.